Hagb/decryptBooxUpdateUpx #1: Strings for Max3; Another method to fetch decrypted update.zip FYI

shunf4, 2020-09-28:
First thanks for your work, it works perfectly on my Max3 with strings below:
```
MODEL = "Max3" 
STRING_7F00500 = "1wdvUHZmcz32N1pgG4fkHmDsTDVihMJsPCNV4mW/6u1k" 
STRING_7F00501 = "3nxuLgdpBE3B3n1Yyymt4cOS8dNucfQxK8YOsmcemuyO" 
STRING_7F00502 = "yCA9YlFxLBdLbDUl3vwzPkn9vtYuVFZCfhrOTvR1" 
```

Just for information purposes, there is another method that has been discussed in the thread [Max3 Rooted](https://www.mobileread.com/forums/showthread.php?t=323254&page=4) in MobileRead Fourms.

To get the decrypted `update.zip` from system upgrade process:
- Ensure you have root permission, preferably by installing Magisk, preferably by booting from a patched `recovery.img` from fastboot (It's always safer that the system and the boot image are not tainted), as per [Magisk installation instruction](https://topjohnwu.github.io/Magisk/install.html)
- In `adb shell`, gain root permission by `su`, then go to `/data/local`
- Duplicate `assets` directory: `cp -a assets assets_2`
- Mount `assets_2` to `assets`: `mount -o bind /data/local/assets_2 /data/local/assets`
- Check for updates, download and install upgrade in settings, just as if you were in a normal upgrading process (If you encountered with some error that says something like "incorrect version", check permissions of the directory above. This is probably because the updater can't access `/data/local/assets`)
- Decrypted `update.zip` is generated in mounted `/data/local/assets`, then the device reboots to recovery and upgrade process hangs, because the decrypted `update.zip` is now missing under `/data/local/assets`, but in `/data/local/assets_2`
- Reboot and with root permission, fetch `/data/local/assets_2/update.zip`. That is the decrypted update data.

Also beware of possibility that all the methods could be invalidated by Onyx. In the thread above they quickly blocked EDL mode after the OP publicly mentioned that.

Hagb, 2020-09-28:
Thanks for your information! I didn't know whether boox use the same algorithm to decrypt `update.zip` until this issue.
And may I mention the issue in `README.md` for this method to get `update.zip`?

shunf4, 2020-09-28:
> And may I mention the issue in `README.md` for this method to get `update.zip`?

Sure

Hagb, 2020-10-16:
The strings are verified.
(By the way, there is a method to get url of boox firmware. https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389)

talamascool, 2021-12-22:
has someone try this method on the new 3.2 update?