Hagb/decryptBooxUpdateUpx #2: Strings for MaxLumi

EnJens, 2020-10-05:
The ID numbers don't match and there are no update.upx's to actually test with yet, but:
```
Model: MaxLumi
(id: 0x7f06000f) settings = mTZFN0K+oMcGnn2n7+zV5DH7kr/Hbes2x/wKDJp6K7Kq
(id: 0x7f060011) upgrade = mj0zR0Oy3L4R+6y49MIEQT9bdx9AVz8TWyG9q3N+d9VY
(id: 0x7f06000a) local = hWAUdhOp9ekIYxIW+LpVj6OviWBbCbRa1c7s1jtW
```
If anyone has any good ideas on how to get an update.upx despite Onyx not releasing them yet it would be appreciated.

Hagb, 2020-10-06:
Thanks for your strings!

> If anyone has any good ideas on how to get an update.upx despite Onyx not releasing them yet it would be appreciated.

The latest firmware downloading url is in `downloadUrlList` in the json from following url:
`http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"NovaPro","submodel":""`, in which
- the domain can be changed to `en-data.onyx-international.cn`
- value of `model` may be the same as `Model` string
- value of `lang` seems about the location of the server, for example `zh_CN` returns the url located in China, and `en_US` returns the url located in US.

For example,`NovaPro`: [http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"NovaPro","submodel":""}](http://en-data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"NovaPro","submodel":""})

And `MaxLumi`:
[http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"MaxLumi","submodel":""}](http://en-data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"MaxLumi","submodel":""})

KOLANICH, 2020-10-06:
https://onyx-boox.ru/firmware is the list of all firmwares for the Onyx readers sold in Russia (but the keys mentioned in the this repo don't fit to them)

Hagb, 2020-10-07:
> https://onyx-boox.ru/firmware is the list of all firmwares for the Onyx readers sold in Russia (but the keys mentioned in the this repo don't fit to them)

@KOLANICH 
Thanks for your information!

sellin82, 2024-01-19:
Note Air 3 model
I would like to download the firmware.

It doesn't download.

http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"NoteAir3","submodel":"","fingerprint":""}

This site address does not work.

miklakt, 2025-01-13:
For NoteAir4C
this request seems to work

http://data.onyx-international.cn/api/firmware/update?where={%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22en_US%22,%22model%22:%22NoteAir4C%22,%22submodel%22:%22%22,%22fingerprint%22:%20%22Onyx/NoteAir4C/NoteAir4C:11/2024-10-23_17-54_4.0_6f06cb790/1287:user/release-keys%22}

raychen501, 2025-01-14:
For BOOX Nova Air C
this request seems to work

http://data.onyx-international.cn/api/firmware/update?where=%7B%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22en_US%22,%22model%22:%22NovaAirC%22,%22submodel%22:%22%22,%22fingerprint%22:%20%22Onyx/NovaAirC/NovaAirC:11/2024-10-23_17-54_4.0_6f06cb790/1287:user/release-keys%22%7D

mio-19, 2025-02-14:
> Thanks for your strings!
> 
> > If anyone has any good ideas on how to get an update.upx despite Onyx not releasing them yet it would be appreciated.
> 
> The latest firmware downloading url is in `downloadUrlList` in the json from following url: `http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"NovaPro","submodel":"","fingerprint":""}`, in which
> 
> * the domain can be changed to `en-data.onyx-international.cn`
> * value of `model` may be the same as `Model` string
> * value of `lang` seems about the location of the server, for example `zh_CN` returns the url located in China, and `en_US` returns the url located in US.
> 
> For example,`NovaPro`: [http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"NovaPro","submodel":"","fingerprint":""}](http://data.onyx-international.cn/api/firmware/update?where=%7B%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22en_US%22,%22model%22:%22NovaPro%22,%22submodel%22:%22%22,%22fingerprint%22:%22%22%7D)
> 
> And `MaxLumi`: [http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"MaxLumi","submodel":"","fingerprint":""}](http://data.onyx-international.cn/api/firmware/update?where=%7B%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22en_US%22,%22model%22:%22MaxLumi%22,%22submodel%22:%22%22,%22fingerprint%22:%22%22%7D)
> 
> Sorry for my bad English.
> 
> 2020-11-28 update: Found that if `"fingerprint"` is not added, server will return `internal server error`.
> 
> 2021-11-26 update: Found that some firmware updates need `"fingerprint"` argument filled, such as `fingerprint="Onyx/MaxLumi2/MaxLumi2:11/2021-11-18_04-40_3.2_6ab9d89216/31254:user/dev-keys"` for MaxLumi2. It is because Onyx now also releases cumulative updates. If all updates of a model are cumulative, the above url without `fingerprint` filled will return `204 No Content`.

How about beta firmwares?

timschneeb, 2025-03-04:
> How about beta firmwares?

I was able to download beta firmware by supplying the device MAC of my reader. The reader assigned to the MAC has to be enrolled in the beta program.

Erichtml, 2025-06-28:
有谁有文石tab8c4.0或4.1的固件吗，感谢。
Has anyone had the version 4.0 or 4.1 firmware of the Boox Tab8C? Thanks so much!

RenateUSB, 2025-06-28:
For the (Chinese) Tab8C there is currently only Onyx/Tab8C/Tab8C:11/2024-09-11_08-17_3.5.4_fbdec6b69a/47380:user/release-keys
There could be a beta, but without a subscribed MAC I wouldn't be able to see it.

Erichtml, 2025-06-28:
So if I have the MAC address of the Tab8C participating in the beta program, do you have a way to extract the system firmware above 4.0?

RenateUSB, 2025-06-28:
You can install it without extracting anything.

AndyCGYan, 2025-07-28:
Since OS 4.0 it seems no longer possible to fetch with the URL method, even if the OS is a release version and I've supplied my MAC and "onyxfp" in the URL. Can someone else verify?

RenateUSB, 2025-07-28:
> I have the Nova Air C and I want to flash the package of Tab8C.

Cross flashing is a separate theme which is best covered elsewhere, like in mobileread.

> Since OS 4.0 it seems no longer possible to fetch with the URL method, even if the OS is a release version and I've supplied my MAC and "onyxfp" in the URL. Can someone else verify?

I'm not sure what you mean or which device. I'm currently DLing NoteAir4C 4.0

AndyCGYan, 2025-07-28:
> I'm not sure what you mean or which device. I'm currently DLing NoteAir4C 4.0

Leaf3, zh_CN.

RenateUSB, 2025-07-28:
This is specific to Leaf3.
Apparently the last non-beta is 3.54

ArchaicBubba, 2025-08-17:
This worked for the Go Color 7/
http://data.onyx-international.cn/api/firmware/update?where={%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22en_US%22,%22model%22:%22GoColor7%22,%22submodel%22:%22%22,%22fingerprint%22:%20%22Onyx/GoColor7/GoColor%22}