Hagb/decryptBooxUpdateUpx #5: Strings for Note3

mgrub, 2020-11-14:
Hey,
first of all, thanks for this great source of information to get insight into our devices!

I have extracted the necessary strings for the new Note3. Here they are:
```python
boox_strings = {
    'Note3': {
        "MODEL": "Note3",
        "STRING_7F00500": "uTiMM5JgTXOCZAZKMcZIzc1yQpfX1+jxTFOred3te4z9",  # settings
        "STRING_7F00501": "zEf3SZ8TOADA8QuwOHicGLrrc4EA7sffKfc01TlUfe/q",  # upgrade
        "STRING_7F00502": "pmXVBMt5EllxXhD9L6/NWH/pTZXRURP6QLsrNlx6"       # local
    }, 
}
```
I was able to decrypt the latest OTA-update.upx with these values, but please check for yourself. 

Note:
The location of the *.apk file was at a different location then suggested in your README. 
But using `pm list packages -f | grep ota` should help you on finding the path.

Hagb, 2020-11-14:
Thanks for your strings and note! And the strings are verified.
May I add your note to README?

mgrub, 2020-11-14:
> Thanks for your strings and note! And the strings are verified.

Great :+1: 

> May I add your note to README?

Yes, feel free to add it.

closed 2020-11-14 by Hagb