Hagb/decryptBooxUpdateUpx #13: Strings for Nova

winstonma, 2021-01-15:
First of all, thanks for this great source of information to get insight into our devices!

I have extracted the necessary strings for the Nova. Here they are:

```
    'Nova': {
        "MODEL": "Nova",
        "STRING_SETTINGS": "L0uopm+jYaWWf/0e/POLt0kkBuS3H+5axpS6cqUpn4ft",
        "STRING_UPGRADE": "XE6jpB3WZ9J5xQdh6GFchFbeBMALt6Zx/UIg8jaiaI72",
        "STRING_LOCAL": "Rh/6kzjOT4nJCsXC5JMEkcbPzzBmNkB8i/c6ZNun"
    }, 
```

I was able to decrypt the latest update.upx with these values, but please check for yourself.

winstonma, 2021-01-15:
Although I could obtain the boot.img, patch the boot.img using Magisk and flash the patched boot.img (using [AndroidTool](https://github.com/rockchip-linux/tools/blob/rk3288/windows/AndroidTool_Release_v2.52.zip)) but the system won't boot up.

I guess the magisk doesn't work well with the Rockchip device.

Hagb, 2021-01-15:
The strings are verifed. Thanks for your issue.

---
I also tried to patch `boot` image for my NovaPro by the tools provided by [rockchip](https://github.com/rockchip-linux), and then got an unbootable image. Even a simply repacked `boot` image without any patch was unbootable in my case. Maybe there were something mistaken.

closed 2021-01-15 by Hagb

winstonma, 2021-01-15:
Yes, actually I also chatted with different people in https://github.com/topjohnwu/Magisk/issues/755. I also tried to use [AIK](https://github.com/draekko/AIK-Linux) to unpack and repack. Got the same result as you. And still have no luck.

On the other hand, do you think it is possible to sideload [SuperSU](https://download.chainfire.eu/1220/SuperSU) in the recovery? I know it is impossible to sideload the zip file directly, but just wonder if it is possible to decrypt the upx, add supersu into the zip file and then encrypt the file back. And then let the system to flash the patched-update.upx for us?

Hagb, 2021-01-15:
I think is impossible to make SuperSU zip work on original recovery since it will verify the sign of zip file.
With an available custom recovery flashed, it is possible to sideload the zip file.

---

Or if a custom recovery is not found, it is also possible to read `system` partition, manually patch it in your computer and rewrite, by [`rkflashtool`](https://github.com/rockchip-linux/rkflashtool). Unlike `boot`, `system` partition is just a normal ext4 filesystem.
I successfully rooted my NovaPro by this method.

winstonma, 2021-01-15:
That's the good news. I know it is off-topic but may I know how could I root my Nova?