Hagb/decryptBooxUpdateUpx #17: Can anyone help to get the string for Dedao Max( Note Air)?

James-Zhu-CA, 2021-07-29:
I am trying to flash the Dedao firmware to my Note air through the update menu, but it didn't allow. If I get the zip
file, maybe there is other way I can flash it in.

I get the file from here https://pan.baidu.com/s/18yacVkce64ecwOi6YA_LXw   code: f2UA

James-Zhu-CA, 2021-07-29:
Or is it possible to modify the upx file to fake a note air version, maybe that also can pass the model verification.

Hagb, 2021-08-05:
> I am trying to flash the Dedao firmware to my Note air through the update menu, but it didn't allow. If I get the zip
> file, maybe there is other way I can flash it in.
> 
> I get the file from here https://pan.baidu.com/s/18yacVkce64ecwOi6YA_LXw code: f2UA

As far as I know, to get the encryption key (called "strings" in this project), firstly a device with the firmware installed already is needed because there is not other way to get `OnyxOtaService.apk` for this firmware.

RenateUSB, 2023-11-27:
This issue is very old. If you are still interested, please reissue it.

closed 2023-11-27 by RenateUSB