Hagb/decryptBooxUpdateUpx #18: Strings for Kon-Tiki 2

mergen3107, 2021-08-07:
Hi! I am trying to decrypt the update.upx for Onyx Boox Kon-Tiki 2. 
There is no package `/system/app/OnyxOtaService/OnyxOtaService.apk`, but there is package `/system/priv-app/OnyxOtaService/OnyxOtaService.apk`. So I got it, apktool decompressed and got the strings from /res/values/strings.xml : 
```
    'KonTiki2': {
        "MODEL": "KonTiki2",
        "STRING_SETTINGS": "eqxOVE1h8e8hbGNiV2ZHed6hMpcOH3vULx6XMm/WguZ1",
        "STRING_UPGRADE": "dKA6Ik0R95nDwjGR/dfPwxkYYNBkfngJk51A2MlRxBsq",
        "STRING_LOCAL": "bPBuYxkL37AYgKbD6nRxEL6EaFCFzFgPij53pTwH"
    }
```

However, when I run the updated `python DeBooxUpx.py KonTiki2 update.upx` with the update file in the same folder, I get this:
```
Traceback (most recent call last):
  File "C:\Users\merge\Documents\!KonTiki2\1 Update\decryptBooxUpdateUpx-master\DeBooxUpx.py", line 169, in <module>
    decrypter = DeBooxUpx(**boox_strings[device_name])
  File "C:\Users\merge\Documents\!KonTiki2\1 Update\decryptBooxUpdateUpx-master\DeBooxUpx.py", line 132, in __init__
    self.decryptStr(tmpKey, STRING_SETTINGS))
  File "C:\Users\merge\Documents\!KonTiki2\1 Update\decryptBooxUpdateUpx-master\DeBooxUpx.py", line 139, in decryptStr
    return cipher.decrypt(b64decode(string)).decode().strip()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9c in position 0: invalid start byte
```

Is there a way to fix this?

The update.upx can be downloaded from here: https://onyx-boox.ru/support/boox_kontiki2

Thank you!

mergen3107, 2021-08-07:
Found the problem!
So the issue was that I used generic `KonTiki2` for the `Model` in the `DeBooxUpx.py` file. The correct way is to do this:
`adb devices -l`
which will give `...model:Kon_Tiki2...` and insert it in the `DeBooxUpx.py`. It turns out that the update.upx expects the correct model name after all. 
The output update.zip opened correctly.

Can you please add these stings to the main page?
Thank you!

mergen3107, 2021-08-07:
Here is how the whole block for Kon-Tiki 2 should look like:
```

    'KonTiki2': {
        "MODEL": "Kon_Tiki2",
        "STRING_SETTINGS": "eqxOVE1h8e8hbGNiV2ZHed6hMpcOH3vULx6XMm/WguZ1",
        "STRING_UPGRADE": "dKA6Ik0R95nDwjGR/dfPwxkYYNBkfngJk51A2MlRxBsq",
        "STRING_LOCAL": "bPBuYxkL37AYgKbD6nRxEL6EaFCFzFgPij53pTwH"
    }
```

Hagb, 2021-08-07:
Hi, thanks for your strings! Now we know that the encryption/decryption algorithm of Russian versions is likely same as other.
And could you please provide `/system/priv-app/kcb-release/oat/arm/kcb-release.odex` and `/system/priv-app/kcb-release/kcb-release.apk`? From them maybe I can find a method to download upx of Russian versions, just like https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389

mergen3107, 2021-08-07:
Here is an archived `/system/priv-app/kcb-release/kcb-release.apk`.
[kcb-release.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/6949559/kcb-release.zip)

For the `odex` file it says it does not exist when I try to adb pull it.

mergen3107, 2021-08-07:
By the way, what does this app do?

Hagb, 2021-08-08:
> By the way, what does this app do?

I don't exactly know much about what it does, but the code fetching upx seems in the app.

Hagb, 2021-08-08:
> Here is an archived `/system/priv-app/kcb-release/kcb-release.apk`.
> [kcb-release.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/6949559/kcb-release.zip)
> 
> For the `odex` file it says it does not exist when I try to adb pull it.

I used to think that the Russian models fetch upx from a different server, but by searching strings in the apk, I find that they likely fetch upx from the same server as international versions (<http://data.onyx-international.cn> and <http://en-data.onyx-international.cn>) with different parameters.

Capturing the network traffic when the e-book checks update may help to get the parameters.

mergen3107, 2021-08-08:
So may I clarify please, to understand how the update works.
Currently I need to:
1) Download the ~1 GB upx file from the website, using PC
2) Put it on the device
3) Go to Settings - check for updates
4) It starts the update if it finds files.

In the process of updating, I can see it opens the upx file and verifies it:
[image]

So is this the step when it fetches MD5 for this file from the server and compares it with that of the user loaded upx file?

> Capturing the network traffic when the e-book checks update

How can I do this? NetGuard detailed logs?

Hagb, 2021-08-08:
>     1. Download the ~1 GB upx file from the website, using PC
> 
>     2. Put it on the device
> 
>     3. Go to Settings - check for updates

In international models, the three steps also work. But they have an system app which can check for updates online (from the update server, described in https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389) and download upx.
I wonder if Russian versions have this function. Or their updates can only be checked and downloaded manually from the web pages?

Usually the latest firmware of international models can only be found in the update server but not onyx web pages where there are only old ones. It is the reason why I am so interested in the method to manually get the firmware from the update server by ourselves.

>     4. It starts the update if it finds files.

These are exactly two steps:
i. An app decrypt the upx to zip, and save it to somewhere (e.g. `/data/local/assets`).
ii. System reboot to recovery, and recovery read the zip and upgrade (just like your photo).

> In the process of updating, I can see it opens the upx file and verifies it:
> 
> So is this the step when it fetches MD5 for this file from the server and compares it with that of the user loaded upx file?

I don't know the answer of this question in step i, but now we don't need to care about it because we can decrypt the upx file by ourselves.
In step ii there are a verification by digital signature and device information indeed.

> > Capturing the network traffic when the e-book checks update

I meant when the e-book checks for updates _online_. (Depend on the previous questions: "I wonder if Russian versions have this function. Or their updates can only be checked and downloaded manually from the web pages?")

> How can I do this? NetGuard detailed logs?

I haven't buy NetGuard pro so I don't know much about it. But it should works.

Sorry for my bad English.

Hagb, 2021-08-12:
> > In international models, the related network traffic is a HTTP GET to `http://data.onyx-international.cn/api/firmware/update?where=SOME_PARAMETERS` or `http://en-data.onyx-international.cn/api/firmware/update?where=SOME_PARAMETERS`.
> 
> @Hagb
> 
> So I used Charles Web Debugging Proxy to see all the requests.
> 
> I had to disable NetGuard for a while. The requests are the same:
> 
> ```
> URL: http://en-data.onyx-international.cn/api/firmware/update?where=....
> Remote address: en-data.onyx-international.cn/47.88.4.191:80
> ```
> 
> I confirm that Russian models knock on the same doors as the international models to check for updates.
> However, if I substitute the model `Kon_Tiki2` as in this URL into the trick with updates, the link does not provide a website with links to download. So I suspect that Russian models do not have OTA updates, but rather only local-PC updates (I need to manually download the upx from PC and transfer to it the device, then click the button on the device to check for updates).

Was there `buildNumber` argument in the part after `where=` of the url? To get information, its value should be modified to a smaller number such as `0`.

mergen3107, 2021-08-12:
Yes, build number is 2389, but even with 0 it does not show anything, even an error page.
Like with this link:
http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"Kon_Tiki2","submodel":"","fingerprint":""}

or this:
http://en-data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"Kon_Tiki2","submodel":"","fingerprint":""}

closed 2021-08-12 by Hagb

mergen3107, 2021-08-12:
Thank you for such a great tool :)