Hagb/decryptBooxUpdateUpx #22: Getting Strings from Max Lumi 2

lazer15, 2021-10-30:
Can the strings be collected using the instructions here on a max lumi 2 without root?
https://github.com/Hagb/decryptBooxUpdateUpx#the-strings
I ordered one but wasn't sure if I'd be able to grab the strings given it isn't rooted yet.

Hagb, 2021-10-31:
https://github.com/Hagb/decryptBooxUpdateUpx#how-to-get-the-strings can be used to get the strings without rooting.

lazer15, 2021-11-01:
I'll use those instructions and attempt to grab and post the strings this weekend when the device arrives.

lazer15, 2021-11-07:
I got the apk but I don't see any keys in the res/values.xml file.

zzfeed, 2021-11-08:
upload your OnyxOtaService.apk file please

kot2002, 2021-11-08:
This is OnyxOtaService.apk/res/values/strings.xml

<?xml version="1.0" encoding="utf-8"?>
<resources>
    <string name="action_settings">Settings</string>
    <string name="app_name">OTA service</string>
    <string name="battery_level_not_safe">Low battery! Please ensure a minimum 30% of the battery for the update.</string>
    <string name="can_not_find_update_package">No update package detected</string>
    <string name="cancel">Cancel</string>
    <string name="disk_io_error">Error in copying update package.</string>
    <string name="disk_space_not_enough">Unused storage is less than 2GB. Please free up some space for the update.</string>
    <string name="failed_to_decompress_package">Failed to decompress upgrade package.</string>
    <string name="find_update_package_from_local">An update package detected from the local storage. Copying package...</string>
    <string name="hello_world">Hello world!</string>
    <string name="is_upgrad">Confirm to update?</string>
    <string name="local" />
    <string name="model_does_not_math">The update package is not for this model. Please download the right update package.</string>
    <string name="ok">OK</string>
    <string name="package_install_falied">Update failed.</string>
    <string name="package_verify_falied">The update package verification failed</string>
    <string name="reboot">Reboot</string>
    <string name="settings" />
    <string name="status_bar_notification_info_overflow">999+</string>
    <string name="upgrad">Update</string>
    <string name="upgrad_successed">The upgrade has been completed and will take effect after restarting. Reboot now?</string>
    <string name="upgrade" />
    <string name="upgrade_package_is_incomplete">The update package is damaged. Please download again.</string>
    <string name="upgrade_unsuccessful">The upgrade unsuccessful! Error :</string>
    <string name="waiting">Updating! Do not power off or turn to sleep mode! The whole update will take a few minutes.</string>
</resources>

[OnyxOtaService.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/7496502/OnyxOtaService.zip)

Hagb, 2021-11-08:
It seems that the method to get keys (in the official application) has been changed to a native (jni) method in `libota_jni.so` binrary file, so our old road is closed in this case now.

---
Could you please run `adb shell 'find /system -name libota_jni.so'` to see whether there is such a file?
If there is any output except for the `Permission denied` error, get the `libota_jni.so` file by `adb pull PATH_TO_FILE` and upload it please.

lazer15, 2021-11-08:
I get inaccessible or not found.
`/system/bin/sh: find /system -name libota_jni.so: inaccessible or not found`

lazer15, 2021-11-08:
Results:
`find: '/system/lib64/bootstrap': Permission denied
/system/priv-app/OnyxOtaService/lib/arm64/libota_jni.so
find: '/system/lib/bootstrap': Permission denied
find: '/system/bin/apexd': Permission denied
find: '/system/bin/audioserver': Permission denied
find: '/system/bin/auditctl': Permission denied
find: '/system/bin/blank_screen': Permission denied
find: '/system/bin/blkid': Permission denied
find: '/system/bin/bootanimation': Permission denied
find: '/system/bin/bootstat': Permission denied
find: '/system/bin/boringssl_self_test32': Permission denied
find: '/system/bin/boringssl_self_test64': Permission denied
find: '/system/bin/bpfloader': Permission denied
find: '/system/bin/cameraserver': Permission denied
find: '/system/bin/charger': Permission denied
find: '/system/bin/clatd': Permission denied
find: '/system/bin/credstore': Permission denied
find: '/system/bin/dex2oat': Permission denied
find: '/system/bin/dnsmasq': Permission denied
find: '/system/bin/drmserver': Permission denied
find: '/system/bin/dumpstate': Permission denied
find: '/system/bin/e2fsck': Permission denied
find: '/system/bin/e2fsdroid': Permission denied
find: '/system/bin/flags_health_check': Permission denied
find: '/system/bin/fsck.f2fs': Permission denied
find: '/system/bin/fsck_msdos': Permission denied
find: '/system/bin/fsverity_init': Permission denied
find: '/system/bin/gatekeeperd': Permission denied
find: '/system/bin/gpuservice': Permission denied
find: '/system/bin/gsid': Permission denied
find: '/system/bin/heapprofd': Permission denied
find: '/system/bin/hw/android.hidl.allocator@1.0-service': Permission denied
find: '/system/bin/hw/android.system.suspend@1.0-service': Permission denied
find: '/system/bin/hwservicemanager': Permission denied
find: '/system/bin/idmap2': Permission denied
find: '/system/bin/idmap2d': Permission denied
find: '/system/bin/incident_helper': Permission denied
find: '/system/bin/incidentd': Permission denied
find: '/system/bin/init': Permission denied
find: '/system/bin/init.onyx.misc.sh': Permission denied
find: '/system/bin/installd': Permission denied
find: '/system/bin/iorap.inode2filename': Permission denied
find: '/system/bin/iorap.prefetcherd': Permission denied
find: '/system/bin/iorapd': Permission denied
find: '/system/bin/keystore': Permission denied
find: '/system/bin/linkerconfig': Permission denied
find: '/system/bin/lmkd': Permission denied
find: '/system/bin/logd': Permission denied
find: '/system/bin/lpdumpd': Permission denied
find: '/system/bin/make_f2fs': Permission denied
find: '/system/bin/mdnsd': Permission denied
find: '/system/bin/mediaextractor': Permission denied
find: '/system/bin/mediametrics': Permission denied
find: '/system/bin/mediaserver': Permission denied
find: '/system/bin/migrate_legacy_obb_data.sh': Permission denied
find: '/system/bin/mke2fs': Permission denied
find: '/system/bin/mtpd': Permission denied
find: '/system/bin/netd': Permission denied
find: '/system/bin/netutils-wrapper-1.0': Permission denied
find: '/system/bin/newfs_msdos': Permission denied
find: '/system/bin/notify_traceur.sh': Permission denied
find: '/system/bin/pppd': Permission denied
find: '/system/bin/racoon': Permission denied
find: '/system/bin/recovery-persist': Permission denied
find: '/system/bin/recovery-refresh': Permission denied
find: '/system/bin/rss_hwm_reset': Permission denied
find: '/system/bin/sdcard': Permission denied
find: '/system/bin/servicemanager': Permission denied
find: '/system/bin/sgdisk': Permission denied
find: '/system/bin/sload_f2fs': Permission denied
find: '/system/bin/snapshotctl': Permission denied
find: '/system/bin/storaged': Permission denied
find: '/system/bin/surfaceflinger': Permission denied
find: '/system/bin/tcpdump': Permission denied
find: '/system/bin/tombstoned': Permission denied
find: '/system/bin/traced': Permission denied
find: '/system/bin/traced_perf': Permission denied
find: '/system/bin/traced_probes': Permission denied
find: '/system/bin/tune2fs': Permission denied
find: '/system/bin/uncrypt': Permission denied
find: '/system/bin/update_engine': Permission denied
find: '/system/bin/update_verifier': Permission denied
find: '/system/bin/usbd': Permission denied
find: '/system/bin/vdc': Permission denied
find: '/system/bin/viewcompiler': Permission denied
find: '/system/bin/vold': Permission denied
find: '/system/bin/vold_prepare_subdirs': Permission denied
find: '/system/bin/wait_for_keymaster': Permission denied
find: '/system/bin/watchdogd': Permission denied
find: '/system/bin/wificond': Permission denied
find: '/system/etc/cgroups.json': Permission denied
find: '/system/etc/selinux/plat_hwservice_contexts': Permission denied
find: '/system/etc/selinux/plat_mac_permissions.xml': Permission denied
find: '/system/apex/com.android.conscrypt/bin/boringssl_self_test64': Permission denied
find: '/system/apex/com.android.conscrypt/bin/boringssl_self_test32': Permission denied
find: '/system/apex/com.android.sdkext/bin/derive_sdk': Permission denied
find: '/system/apex/com.android.media.swcodec/bin/mediaswcodec': Permission denied
find: '/system/apex/com.android.os.statsd/bin/statsd': Permission denied
find: '/system/apex/com.android.adbd/bin/adbd': Permission denied
find: '/system/apex/com.android.art.release/bin/dex2oat32': Permission denied
find: '/system/apex/com.android.art.release/bin/profman': Permission denied
find: '/system/apex/com.android.art.release/bin/dexoptanalyzer': Permission denied
find: '/system/apex/com.android.art.release/bin/dex2oat64': Permission denied`

kot2002, 2021-11-08:
[libota_jni.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/7498056/libota_jni.zip)

lazer15, 2021-11-08:
`find: '/system/bin/iorap.inode2filename': Permission denied

find: '/system/bin/iorap.prefetcherd': Permission denied

find: '/system/bin/iorapd': Permission denied`

Hagb, 2021-11-09:
TL;DR It seems that the strings are:
```
STRING_SETTINGS: zR5/dfK/2XhmwhA9AIGr+gx6Vzho5sbklSTjLAgnC8/C
STRING_UPGRADE: zh98dPzN0QdwbRem1Yfx28DFxU+Gkvg4MpLDIp0GorRU
```
(I am not completely sure)

---

What I did:
1. Install `openjdk-17` package in [Termux](https://github.com/termux/termux-app). (It is too heavy. It costs hundreds of MB.)
2. Write a java file and save it in `./com/onyx/android/onyxotaservice/RsaUtil.java`, which just does what the real OnyxOtaService does on `libota_jni.so`, except that `libota_jni.so` is specified by argument:
    ```java
    package com.onyx.android.onyxotaservice;
    import java.nio.file.Paths;

    public class RsaUtil {
        public static void main(String args[]){
            if (args.length != 1){
                    System.out.println("Argument: path to libota_jni.so");
                    System.exit(1);
            }
            System.load(Paths.get(args[0]).toAbsolutePath().toString());
            System.out.println("STRING_SETTINGS: " + nativeGetSecretKey());
            System.out.println("STRING_UPGRADE: " + nativeGetIvParameter());
        }
        public static native String nativeGetIvParameter(); # it is implemented in `libota_jni.so` by Onyx
        public static native String nativeGetSecretKey();     # it is implemented in `libota_jni.so` by Onyx
    }
    ```
3. Compile the file
   ```bash
   javac ./com/onyx/android/onyxotaservice/RsaUtil.java` 
   ```
4. Run the class file with the path to `libota_jni.so` as the argument
   ```bash
   java com.onyx.android.onyxotaservice.RsaUtil ~/libota_jni.so
   ```
   In my case the path is `~/libota_jni.so` because my device haven't `libota_jni.so` originally in fact. In the device using `libota_jni.so` the path may be `/system/priv-app/OnyxOtaService/lib/arm64/libota_jni.so`.

It is possible to write an apk to do these things, using just Android jni without the heavy openjdk.

zzfeed, 2021-11-09:
> [libota_jni.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/7498056/libota_jni.zip)

Can you extract kreader apk and upload? thanks in advance

Hagb, 2021-11-25:
Could anyone please paste the output of `adb shell "getprop  ro.vendor.build.fingerprint"`?
It is found that, to get the firmware information, the `fingerprint` argument in https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389 should be filled.

kot2002, 2021-11-26:
ro.system.build.fingerprint=qti/qssi/qssi:11/2021-11-22_12-05_3.2_0ae74d7d5d/31272:user/dev-keys

kot2002, 2021-11-26:
Onyx/MaxLumi2/MaxLumi2:11/2021-11-22_12-05_3.2_0ae74d7d5d/31272:user/dev-keys

Hagb, 2021-11-26:
Now the update file can be fetched from <http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"MaxLumi2","submodel":"","fingerprint":"Onyx/MaxLumi2/MaxLumi2:11/2021-11-18_04-40_3.2_6ab9d89216/31254:user/dev-keys"}>.

```python
"MaxLumi2": {
    "MODEL": "MaxLumi2",
    "STRING_SETTINGS": "zR5/dfK/2XhmwhA9AIGr+gx6Vzho5sbklSTjLAgnC8/C",
    "STRING_UPGRADE": "zh98dPzN0QdwbRem1Yfx28DFxU+Gkvg4MpLDIp0GorRU"
}
```
The keys above, which can decrypted the update.upx , are now verified to be correct. And it also seems that `STRING_LOCAL` doesn't exist anymore.

peasca, 2021-12-11:
> Thank @kot2002 ! Now one of the update files, which are cumulative, can be fetched from [http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"zh_CN","model":"MaxLumi2","submodel":"","fingerprint":"Onyx/MaxLumi2/MaxLumi2:11/2021-11-18_04-40_3.2_6ab9d89216/31254:user/dev-keys"}](http://data.onyx-international.cn/api/firmware/update?where=%7B%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22zh_CN%22,%22model%22:%22MaxLumi2%22,%22submodel%22:%22%22,%22fingerprint%22:%22Onyx/MaxLumi2/MaxLumi2:11/2021-11-18_04-40_3.2_6ab9d89216/31254:user/dev-keys%22%7D).
> 
> ```python
> "MaxLumi2": {
>     "MODEL": "MaxLumi2",
>     "STRING_SETTINGS": "zR5/dfK/2XhmwhA9AIGr+gx6Vzho5sbklSTjLAgnC8/C",
>     "STRING_UPGRADE": "zh98dPzN0QdwbRem1Yfx28DFxU+Gkvg4MpLDIp0GorRU"
> }
> ```
> 
> The keys above, which can decrypted the update.upx , are now verified to be correct. And it also seems that `STRING_LOCAL` doesn't exist anymore.

Can you get the string from my Boox NoteX?
That is the files from the device
[https://drive.google.com/file/d/1Z4HrToCCA1spE7Ci62V12fUKrxPBkprO/view?usp=sharing](url)

Hagb, 2021-12-11:
> Can you get the string from my Boox NoteX? That is the files from the device [https://drive.google.com/file/d/1Z4HrToCCA1spE7Ci62V12fUKrxPBkprO/view?usp=sharing](url)

@peasca  I have no permission to access this file.

peasca, 2021-12-12:
> https://drive.google.com/file/d/1Z4HrToCCA1spE7Ci62V12fUKrxPBkprO/view?usp=sharing

[https://1drv.ms/u/s!AhSpFc557rlvlUYLSMKlJ_uFRbbl?e=XFYWNX](url)
New address

Hagb, 2021-12-12:
> > https://drive.google.com/file/d/1Z4HrToCCA1spE7Ci62V12fUKrxPBkprO/view?usp=sharing
> 
> [https://1drv.ms/u/s!AhSpFc557rlvlUYLSMKlJ_uFRbbl?e=XFYWNX](url) New address

@peasca  I still can't get this file

I just wrote a [app to get the keys](https://github.com/Hagb/GetBooxUpxKeysApp/releases/download/v0.1), and you can try it.

If there are any problem, feel free to tell me.

kot2002, 2021-12-12:
[NoteX.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/7698388/NoteX.zip)

Here it is..

Hagb, 2021-12-12:
> [NoteX.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/7698388/NoteX.zip)
> 
> Here it is..

Thanks! The above key strings is verified:
```python
    "NoteX": {
        "MODEL": "NoteX",
        "STRING_SETTINGS": "xhffOONo9wDGZquv93yJoLm0z6igW+XL17PUyfR8ky4l",
        "STRING_UPGRADE": "zBGvTe9t8HZQXu2Q85Dl0iI9R9tEg2VRw/8cEYXauU3o"
     }
```
---

Could anyone please test https://github.com/Hagb/GetBooxUpxKeysApp if Boox OS of your device is Android 11.
I don't have such a device so I tested the the app with some hacks.

peasca, 2021-12-13:
> > [NoteX.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/7698388/NoteX.zip)
> > Here it is..
> 
> Thanks! The above key strings was got and verified:
> 
> ```python
>     "NoteX": {
>         "MODEL": "NoteX",
>         "STRING_SETTINGS": "xhffOONo9wDGZquv93yJoLm0z6igW+XL17PUyfR8ky4l",
>         "STRING_UPGRADE": "zBGvTe9t8HZQXu2Q85Dl0iI9R9tEg2VRw/8cEYXauU3o"
>      }
> ```
> 
> Could anyone please test https://github.com/Hagb/GetBooxUpxKeysApp if Boox OS of your device is Android 11? I don't have such a device so I tested the the app with some hacks in my Note3 with Android 10.

This tool can get key from BooxNoteX Version:2021/11/11
Thanks！

closed 2021-12-13 by Hagb