Hagb/decryptBooxUpdateUpx #24: 增加设备 SP_NoteS

taoHC1997, 2021-11-20:
我有一台超星Note SL ，是 note S 的定制版

此版本使用： `http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"zh_CN","model":"SP_NoteS","submodel":"","fingerprint":""} `

下面是拆包得到的文件：

````
    'SP_NoteS': {
        "MODEL": "SP_NoteS",
        "STRING_SETTINGS": "XjzywnKKh3FNVKphg5MZ9xILmDNYBZiCHTnkd6Q0fqpq",
        "STRING_UPGRADE": "XDeIyQKAhQKj6/UmdAIWoL0aXmw9PeEFl9OpII/1TO9i",
        "STRING_LOCAL": "QmGqhFGXqiuQ9yoeVx0a2SQ8BNojjPn96o6hQW6U"
    },
````

Hagb, 2021-11-25:
感谢！验证了一下确实无误，加到代码里了。
> Thanks! The strings are verified and merged.

closed 2021-11-25 by Hagb

xif12jhj, 2025-10-31:
请问你的改完有电磁屏固件吗？

RenateUSB, 2025-10-31:
> 请问你的改完有电磁屏固件吗？
> Do you have the electromagnetic shield firmware after you made the changes?

Sorry, I have no idea what you mean.