Hagb/decryptBooxUpdateUpx #30: [Errno 2] No such file or directory

zenminimalist, 2022-03-03:
After this command:

>python DeBooxUpx.py NoteAir2 update.upx

I get the following error message: 
`/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python: can't open file 'DeBooxUpx.py': [Errno 2] No such file or directory`

Any idea what the issue could be?

zenminimalist, 2022-03-03:
I did manage to get the [ibota_jni.so](https://www.dropbox.com/s/tli5wayj5x9avgp/libota_jni.so?dl=0) file, but cannot open it anyway. Not a dev either.

zenminimalist, 2022-03-04:
Just FYI. The Note Air 2 does have different Processors and managed to use this prepatched boot.img for Note Air 2 [here](https://www.mobileread.com/forums/showpost.php?p=4187479&postcount=52).

Hagb, 2022-03-05:
> Any idea what the issue could be? I do get a ZIP file, but with 0 bytes.

As the error output shows, your update.upx seems not for NoteAir2.

> : ( I tried getting the strings via various methods (https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389) without luck...

Do you mean getting the update.upx? Since Android 9 or Android 10, it seems that Onyx no longer provides full updates, and just provides incremental updates, so for these model, `fingerprint` is necessary for fetching the update.upx.

> I did manage to get the [ibota_jni.so](https://www.dropbox.com/s/tli5wayj5x9avgp/libota_jni.so?dl=0) file, but cannot open it anyway. Not a dev either.

https://github.com/Hagb/GetBooxUpxKeysApp/releases can be used ro get key strings from `libota_jni.so`.

zenminimalist, 2022-03-05:
> Do you mean getting the update.upx? Since Android 9 or Android 10, it seems that Onyx no longer provides full updates, and just provides incremental updates, so for these model, `fingerprint` is necessary for fetching the update.upx.
> 
For some reason, this method did not work for me even with the fingerprint.
> 
> https://github.com/Hagb/GetBooxUpxKeysApp/releases can be used to get key strings from `libota_jni.so`.

As for the GetBooxUpxKeysApp, I don't know how to run it How do I make an apk out of it to run it on my BOOX? Sorry for the stupid question, not a dev with deep technical understanding, unfortunately.

zenminimalist, 2022-03-06:
However, I forgot to note down my fingerprint. How do I get it again?

Hagb, 2022-03-06:
> @Hagb For some reason, this method did not work for me even with the fingerprint. But just I'll gladly try again.

The method work only if your fingerprint is not the fingerprint of the latest version. You can get one in https://github.com/Hagb/decryptBooxUpdateUpx/pull/27#issuecomment-1020803415, but it is an _incremental update_, in other words, there are only the different parts between the two versions but not the full system image in this update package, so it is probable that there isn't what you need.

> As for the GetBooxUpxKeysApp, I don't know how to run it How do I make an apk out of it to run it on my BOOX? Sorry for the stupid question, not a dev with deep technical understanding, unfortunately. Can you briefly elaborate on how to do that?

Open the url in your BOOX device, download the apk, and install it as an application.
But I think the strings will be the same with other NoteAir2's.

zenminimalist, 2022-03-06:
> The method work only if your fingerprint is not the fingerprint of the latest version. You can get one in [#27 (comment)](https://github.com/Hagb/decryptBooxUpdateUpx/pull/27#issuecomment-1020803415), but it is an _incremental update_, in other words, there are only the different parts between the two versions but not the full system image in this update package, so it is probable that there isn't what you need.

OK, thanks. 👍🏼 I've not got my fingerprint. 

> Open the url in your BOOX device, download the apk, and install it as an application. But I think the strings will be the same with other NoteAir2(s)'.

OK, now I feel stupid. I didn't see the apk on the main readme. It was under assets. My bad, 对不起 🤦🏼

Just to clarify for my better understanding: 

- I'm currently on 3.2 and have root. I used this [guide](https://blog.tho.ms/hacks/2021/03/27/hacking-onyx-boox-note-air.html) and used a pre-patched version and applied it via magisk so that I can still receive firmware updates on the devices after rooting, by simply patching the boot.img and not lose any user data. (The guide mentioned: _"Because this is a systemless root, you will still be able to update the firmware through their update mechanism. However, you will lose root and have to re-root afterwards._")
- So, if the upcoming version 3.2.1 comes out, we need a **full** firmware in order to extract the boot.img and to be able to root the version 3.2.1, correct? Or can you just update OTA via the device to 3.2.1 and use the previous patched boot.img? Or will the boot.img be different from version to version?

Hagb, 2022-03-06:
> So, if the upcoming version 3.2.1 comes out, we need a full firmware in order to extract the boot.img and to be able to root the version 3.2.1, correct? Or can you just update OTA via the device to [...]

If the `boot.img` be different from version A to version B, then the newer `boot.img` will in the `update.upx` used to upgrade from version A to version B, so the boot.img can be got from this `boot.img`, and then patched by magisk, and then flashed to the device. BTW, to get this `update.upx` you can use the method in https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389 _with the fingerprint of version A_.

If the `boot.img` be the same between version A and version B (in other words, the `boot.img` be not different between version A and version B), you can just apply the patched boot.img (now you have got it in [the guide](https://blog.tho.ms/hacks/2021/03/27/hacking-onyx-boox-note-air.html)) used in version A, after updating to version B.

And if you have any questions just feel free to ask. :-)

zenminimalist, 2022-03-06:
Ok, thanks, That clears things up quite a bit!

closed 2022-03-06 by zenminimalist

zenminimalist, 2022-03-09:
Using my current fingerprint (rooted on v3.2) I could download the following update 3.2.1.

[Incremental update](http://firmware-us.boox.com/e60d3109279fc90989d86f9f493d98f8/update.upx)
This download is about 190 MB, which makes sense since it's a delta/incremental update.

I also downloaded this update from [here](https://github.com/Hagb/decryptBooxUpdateUpx/pull/27#issuecomment-1020803415). I get a 1.2 GB file.

I both used the following command `python3 DeBooxUpx.py NoteAir2 update.upx` to decrypt both update files to compare them. And in both update files I have [this](https://imgur.com/a/iagwjsL) (same) directory structure. 

I'm missing [this](https://imgur.com/a/BFHOhdJ) directory structure. There's no boot.img. Am I missing something?

Don't I need the boot.img from the new v3.2.1 first? 

Or do I:
1. Update from 3.2 --> 3.2.1 and lose my root
2. Pull the boot.img from the device that is now on 3.2.1
3. Patch it via Magisk?
4. Have root?

Hagb, 2022-03-16:
> I also downloaded this update from [here](https://github.com/Hagb/decryptBooxUpdateUpx/pull/27#issuecomment-1020803415). I get a 1.2 GB file.
>
> I both used the following command `python3 DeBooxUpx.py [...]

It seems an incremental update.

> @Hagb Don't I need the boot.img from the new v3.2.1 first?

Don't need.
 
> Or do I:
> 
>     1. Update from 3.2 --> 3.2.1 and lose my root
>
>     2. Pull the boot.img from the device that is now on 3.2.1
> 
>     3. Patch it via Magisk?
> 
>     4. Have root?

Try to flash your old boot.img patched by magisk (don't need to patch it again) after updating from 3.2 to 3.2.1, and root should be there.