Hagb/decryptBooxUpdateUpx #32: Strings for Nova Air C

mcoombs32, 2022-04-01:
I just got the Nova Air C and pulled the strings:
`    'NovaAirC': {
        "MODEL": "NovaAirC",
        "STRING_SETTINGS": "ycihhTCxjfQO85KdVPvpA1ZzCPr3eJ/KhcKEKlqMhEfR",
        "STRING_UPGRADE": "zsnQgzXEhvsfL4pk48rtEckIPyDedhybURHP198tag+Z"
    }`

No STRING_LOCAL, and I'm getting an error when trying to decrypt, but these are copy/pasted straight from Get Boox Upx Keys:
\Downloads\decryptBooxUpdateUpx-master\decryptBooxUpdateUpx-master\DeBooxUpx.py", line 206, in deUpxSteam
    raise ValueError("The decrypted data seems not a zip package, "
ValueError: The decrypted data seems not a zip package, please ensure that the strings or model is correct.

mcoombs32, 2022-04-01:
Just kidding, i had the .upx for the Nova 3 Color instead of the Nova Air C. Will update once i find the right .upx

Hagb, 2022-04-05:
Thanks for your strings! Merged.

closed 2022-04-05 by Hagb