Hagb/decryptBooxUpdateUpx #62: Tab Ultra strings

RenateUSB, 2023-08-24:
'TabUltra': {
        'MODEL': 'TabUltra',
        'STRING_SETTINGS': 'zsb+9c4SzqrAnPkdtIaoKcw4r5gjLr+anKJDYlypaMR+',
        'STRING_UPGRADE': 'yrX2h8weyduHDTK+oWbibAXbqBjGbDyEjeDb1pRRmkIl'
    }

usb9245, 2023-11-21:
```python3
'Poke5': {
    "MODEL": "Poke5",
    "STRING_SETTINGS": "WUg6pwei2Auz1Bt3KTw/VvYzYLyahqYmoRrp0CNDpQgw",
    "STRING_UPGRADE": "WDI+rnel2ni3FyHk8KtD/2NZcHit0b0ptb4k+QqAieKw"
}
```

for Poke5

RenateUSB, 2023-11-21:
What is the difference between a Poke5 and a Poke5P?
I bought what was advertised as "Poke5", but the model and fingerprint show "Poke5P".
I've also not seen any updates for the "Poke5"

Also, Python is not JSON, we can just use single quotes everywhere.

usb9245, 2023-11-21:
I think "Poke5" is chinese version of Poke5.
http://data.onyx-international.cn/api/firmware/update?where={%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22zh_CN%22,%22model%22:%22Poke5%22,%22submodel%22:%22%22,%22fingerprint%22:%22%22}
[image]

RenateUSB, 2023-11-21:
Oh! Thank you. The language is zh_CN:
```
Model:  Poke5
Finger: Onyx/Poke5/Poke5:11/2023-11-17_04-09_3.5_801526b7d/6777:user/dev-keys
Link:   http://firmware-volc.boox.com/44207d5ef900c54aeab756b596c29354/update.upx
Size:   1595705515
MD5:    44207d5ef900c54aeab756b596c29354
```

RenateUSB, 2023-11-22:
@usb9245 Pasting that as an image is too error-prone. Could you paste that as text or else post /system/lib64/libota_jni.so ? Tnx!

Hagb, 2023-11-22:
Oh interesting. In the case of Leaf2, `Leaf2_P` is Chinese version and `Leaf2` is international version.

usb9245, 2023-11-22:
> @usb9245 Pasting that as an image is too error-prone. Could you paste that as text or else post /system/lib64/libota_jni.so ? Tnx!

[libota_jni.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/13441934/libota_jni.zip)

Iey4iej3, 2023-11-24:
Close this issue by https://github.com/Hagb/decryptBooxUpdateUpx/commit/d48b8cbc5cbaab4339840462da77ed7657d2357d

closed 2023-11-24 by RenateUSB