Hagb/decryptBooxUpdateUpx #66: Tab10CPro Strings

junrenshi, 2023-11-04:
I install GetBooxUpxKeys in a Boox Tab 10C Pro, and get outputs:

> "MODEL" = "Tab10CPro",
> "STRING_SETTINGS" = "D999393AEAA81119AC0F8C8C1EA11089",
> "STRING_UPGRADE" = "CD7894A509490BAF2BAA129686A083E4"
>  # fingerprint: Onyx/Tab10CPro/Tab10CPro:11/2023-10-30_17-05_3.5_edb1c449c/4119:user/release-keys

Using the strings to decrypt update.upx results in error:

> > python3 DeBooxUpx.py Tab10CPro
> Traceback (most recent call last):
>   File "/Users/shi/Downloads/decryptBooxUpdateUpx/DeBooxUpx.py", line 337, in <module>
>     decrypter = DeBooxUpx(**boox_strings[device_name])
>                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/shi/Downloads/decryptBooxUpdateUpx/DeBooxUpx.py", line 289, in __init__
>     self.decryptStr(tmpKey, STRING_SETTINGS))
>     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/shi/Downloads/decryptBooxUpdateUpx/DeBooxUpx.py", line 296, in decryptStr
>     return cipher.decrypt(b64decode(string)).decode().strip()
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> UnicodeDecodeError: 'utf-8' codec can't decode bytes in position 0-1: invalid continuation byte
> 

It seems GetBooxUpxKeys cannot get correct strings for this new device.

RenateUSB, 2023-11-07:
Post libota_jni.so and I'll check without Java.

junrenshi, 2023-11-08:
Yes, they do look like hexadecimal numbers.

 libota_ini.so is here:
[libota_jni.tgz](https://github.com/Hagb/decryptBooxUpdateUpx/files/13302056/libota_jni.tgz)

RenateUSB, 2023-11-09:
I get the same results as you.
(The libjni_ota.so does not have those values as either binary or string directly in it. It's generated.)
But that is 32 hex digits, which is 16 bytes or 128 bits.

The strings we've been using have been 44 characters which can be evenly base64 converted to 33 bytes or 264 bits.
(Note that none of the strings end in '=', which means it's an even 4 char to 3 byte conversion.)

This new libota_jni.so has a third JNI function, Java_com_onyx_android_onyxotaservice_RsaUtil_nativeDecryptFile
So maybe things have changed radically. @Hagb

Hagb, 2023-11-09:
@junrenshi Could you please upload your `/system/app/OnyxOtaService/OnyxOtaService.apk`? It might be needed to decompile this apk to try to find how it works.

junrenshi, 2023-11-09:
The file is now moved to /sys/priv-app/OnyxOtaService/.

Here is the file:
[OnyxOtaService.apk.tgz](https://github.com/Hagb/decryptBooxUpdateUpx/files/13303685/OnyxOtaService.apk.tgz)

RenateUSB, 2023-11-09:
Would you have a link for that update.upx?
I can't seem to find it.

junrenshi, 2023-11-09:
Sure.  It is http://firmware-volc.boox.com/ff9cd4c9b96543cee0e77b097f9949a6/update.upx

RenateUSB, 2023-11-09:
Works nicely for me. :)
```
Poke3:/data/local/assets # unzip -l update.zip
Archive:  update.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
      155  2009-01-01 00:00   payload_properties.txt
        0  2009-01-01 00:00   apex_info.pb
      678  2009-01-01 00:00   care_map.pb
      615  2009-01-01 00:00   META-INF/com/android/metadata
     1030  2009-01-01 00:00   META-INF/com/android/metadata.pb
     1513  2009-01-01 00:00   META-INF/com/android/otacert
1556748068  2009-01-01 00:00   payload.bin
---------                     -------
1556752059                     7 files
```

junrenshi, 2023-11-10:
Great! You did that.  Can I do that as well?

RenateUSB, 2023-11-10:
That was my proof of concept, extracting without Python or Java.
I don't pretend to understand any of this encryption junk.

You can mod the stock DeBooxUpd to only work for the Tab10CPro.
@Hagb will have to figure out what to do going forward.

Just insert these two lines in DeBooxUpd, line 287, right after self.path: str =
```
        self.key = b'\xd9\x99\x39\x3a\xea\xa8\x11\x19\xac\x0f\x8c\x8c\x1e\xa1\x10\x89'
        self.iv = b'\xcd\x78\x94\xa5\x09\x49\x0b\xaf\x2b\xaa\x12\x96\x86\xa0\x83\xe4'
```

junrenshi, 2023-11-10:
Great! It also works for me.   Thanks a lot!

hexomethyl, 2023-11-10:
Hey guys,

Recently got my Note air 3C, after looking at both `libota_jni.so` and `OnyxOtaService.apk`, Boox seems to have changed how the decryption of the Upx files work. 

Previously there was a function which returns the AES secret key (`Java_com_onyx_android_onyxotaservice_RsaUtil_nativeGetSecretKey`) and AES IV (`Java_com_onyx_android_onyxotaservice_RsaUtil_nativeGetIvParameter`), these were encrypted using DES (With the model name as the key, lol) and encoded using Base64.

Now for my new device with `fingerprint: "Onyx/NoteAir3C/NoteAir3C:11/2023-10-17_09-55_3.5_a37b266eb/2482:user/test-keys"`, this is no longer the case.

Now there are two functions exported from the libota_jni.so dynamic library that return the direct AES secret key and AES IV. 
These functions have mangled C++ names but are `getKeyString` and `getInitVectorString`

This means that the fancy DES decryption is no longer needed but rather the parameters can be feed straight into AES to perform the decryption.

Here is a modified version of the script to extract the new private key/IV
@Hagb Feel free to use this code

```

from ExAndroidNativeEmu.androidemu.const import emu_const
from ExAndroidNativeEmu.androidemu.emulator import Emulator
from ExAndroidNativeEmu.androidemu.internal import elf_reader
from ExAndroidNativeEmu.androidemu.java.classes.types import *
from ExAndroidNativeEmu.androidemu.utils.memory_helpers import read_utf8

AES_SECRET_KEY_FUNCTION = 'getKeyString'
AES_IV_FUNCTION = 'getInitVectorString'

def call_function(so_path, function_name, *args):
    # Initialize emulator
    reader = elf_reader.ELFReader(so_path)
    if reader.is_elf32():
        emulator = Emulator(
            vfs_root="ExAndroidNativeEmu/vfs",
            arch=emu_const.ARCH_ARM32,
            config_path="ExAndroidNativeEmu/emu_cfg/default.json"
        )
    else:
        emulator = Emulator(
            vfs_root="ExAndroidNativeEmu/vfs",
            arch=emu_const.ARCH_ARM64,
            config_path="ExAndroidNativeEmu/emu_cfg/default.json"
        )

    module = emulator.load_library(so_path)
    function_symbol = [module.find_symbol(export) for export in module.symbols if function_name in export]
    result = emulator.call_native(function_symbol[-1], *args)
    return read_utf8(emulator.mu, result)

if __name__ == '__main__':
    if len(sys.argv) < 2:
        print("python %s /path/to/libota_jni.so" % __file__)
        exit(0)

    secret_key = call_function(sys.argv[1], AES_SECRET_KEY_FUNCTION)
    iv = call_function(sys.argv[1], AES_IV_FUNCTION)

    print(f"AES Secret key: {secret_key}")
    print(f"AES Initialization vector: {iv}")

```

Below I have attached the corresponding binaries and screenshots of the new apk and library functions.

[image]
[image]
[image]

[Binaries.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/13316148/Binaries.zip)

RenateUSB, 2023-11-10:
For now we can just backtrack to the strings:
```
    'Tab10CPro': {
        'MODEL': 'Tab10CPro',
        'STRING_SETTINGS': 'on8l4AwDwIgVNi6AgrAqLzkc94XMx9O37CCELz23Fm+J',
        'STRING_UPGRADE': 'pQIr4QYOsvxLqxXik9cIIzca+QzC2pa9U/hY8p/EAkur'
    }
```

Iey4iej3, 2023-11-22:
> ``` Python
>
> from ExAndroidNativeEmu.androidemu.const import emu_const
> from ExAndroidNativeEmu.androidemu.emulator import Emulator
> from ExAndroidNativeEmu.androidemu.internal import elf_reader
> from ExAndroidNativeEmu.androidemu.java.classes.types import *
> from ExAndroidNativeEmu.androidemu.utils.memory_helpers import read_utf8
> 
> 
> AES_SECRET_KEY_FUNCTION = 'getKeyString'
> AES_IV_FUNCTION = 'getInitVectorString'
> 
> 
> def call_function(so_path, function_name, *args):
>     # Initialize emulator
>     reader = elf_reader.ELFReader(so_path)
>     if reader.is_elf32():
>         emulator = Emulator(
>             vfs_root="ExAndroidNativeEmu/vfs",
>             arch=emu_const.ARCH_ARM32,
>             config_path="ExAndroidNativeEmu/emu_cfg/default.json"
>         )
>     else:
>         emulator = Emulator(
>             vfs_root="ExAndroidNativeEmu/vfs",
>             arch=emu_const.ARCH_ARM64,
>             config_path="ExAndroidNativeEmu/emu_cfg/default.json"
>         )
> 
>     module = emulator.load_library(so_path)
>     function_symbol = [module.find_symbol(export) for export in module.symbols if function_name in export]
>     result = emulator.call_native(function_symbol[-1], *args)
>     return read_utf8(emulator.mu, result)
> 
> 
> if __name__ == '__main__':
>     if len(sys.argv) < 2:
>         print("python %s /path/to/libota_jni.so" % __file__)
>         exit(0)
> 
>     secret_key = call_function(sys.argv[1], AES_SECRET_KEY_FUNCTION)
>     iv = call_function(sys.argv[1], AES_IV_FUNCTION)
> 
>     print(f"AES Secret key: {secret_key}")
>     print(f"AES Initialization vector: {iv}")
> ```

This works perfectly on Note Air 3C, and the key I got is

```
AES Secret key: 4DEAED7F843CDCEDB384E6FC468C540E
AES Initialization vector: 964E8856F922178810AEC06E0D363512
```

I modified the sample code to

``` Python
#!/usr/bin/env python3

from DeBooxUpx import DeBooxUpx

MODEL = "NovaPro" 
STRING_SETTINGS = "j857wYAQcWZgvIEQ/tcQqzxreUJgFHwJl6D2TN9BuSkQ" 
STRING_UPGRADE = "+soGw/YVdGIRJiAs5SMmv1ihW37H1Fa9+/1w2Vgt14Ag" 
STRING_LOCAL = "lpsj9NJ8Kzv8jHb+OO8A5lxC+9Zhl243bFmDZYaF" 

import sys

# Run with
# $ ./decrypt.py INPUT OUTPUT

updateUpxPath = sys.argv[1]
decryptedPath = sys.argv[2]

decrypter = DeBooxUpx(MODEL, STRING_SETTINGS, STRING_UPGRADE, STRING_LOCAL)
decrypter.key = bytes.fromhex("4DEAED7F843CDCEDB384E6FC468C540E")
decrypter.iv = bytes.fromhex("964E8856F922178810AEC06E0D363512")
print('When updating, the device decrypt update package into', decrypter.path)
decrypter.deUpx(updateUpxPath, decryptedPath)
```

and it decrypts the UPX file successfully.

Hagb, 2023-11-22:
Thank @RenateUSB and @hexomethyl ! I'm thinking that we could store these strings in `boox_strings` dict of `DeBooxUpx` like the older ones, and determine whether it is the AES keys or the DES-encrypted AES keys based on the length of the string?

RenateUSB, 2023-11-22:
@Hagb Here you go, it's all done for you:
```
#!/usr/bin/env python3
try:
    from Cryptodome.Cipher import AES
except ModuleNotFoundError:
    from Crypto.Cipher import AES
    from Crypto import version_info
    if version_info[0] == 2:
        raise SystemExit('Need either `pycryptodome` or `pycryptodomex`,'\
                ' NOT `pycrypto`!')

boox_strings = {
    'FLOW': {
        'MODEL': 'FLOW',
        'KEY': '42F6E69DB0A176DC5FBA04DE2C4C0C7D',
        'IV': '9C0F30369B5539A1E0696598638D4731'
    },
    'Gulliver-ru': {
        'MODEL': 'MC_GULLIVER',
        'KEY': '0CBE6C03F589353182354085D8E20F31',
        'IV': '7E34FC7B6E6FD0A39C0E43CBAF32FB52'
    },
    'KonTiki2-ru': {
        'MODEL': 'Kon_Tiki2',
        'KEY': '98AC5EB0D8D7A0A0B2E75B278DB8B288',
        'IV': '745555DF150183D4C92AFBC4EDE58ED2'
    },
    'Leaf': {
        'MODEL': 'Leaf',
        'KEY': '2FA5F9484C5079FA25B89FD6A926F0FD',
        'IV': '0CB5EBC30E284C71108ACC344026DFD6'
    },
    'Leaf2': {
        'MODEL': 'Leaf2',
        'KEY': '4131D58DC6CB5D85BFD23D9F576CAB16',
        'IV': 'A60D8B8DEE2A494194F1AC6C80E850AC'
    },
    'Leaf2-cn': {
        'MODEL': 'Leaf2_P',
        'KEY': 'E95EE8726EC4B63A2241F38829C27FCC',
        'IV': '8093A741A8A4C45E11690B72D78C59A1'
    },
    'Leaf3': {
        'MODEL': 'Leaf3',
        'KEY': '6ABE99B43B928F351C83CDD5A2516A6E',
        'IV': '1009E8860F53BB40483A83173F844198'
    },
    'Lomonosov-ru': {
        'MODEL': 'Lomonosov',
        'KEY': 'D2F7E3354FC07C2521BF609D01FBC90F',
        'IV': '0070D3F87818CF23DB5EE9798B8B0657'
    },
    'Max2': {
        'MODEL': 'Max2',
        'KEY': '7B71BD797D7E3721D69F4449E12B4C22',
        'IV': '38B2BB851DAF20E1BE6E7E3F349CABA0'
    },
    'Max3': {
        'MODEL': 'Max3',
        'KEY': '0C3FF83E57334391A2D4171E99EAC1A9',
        'IV': '982877D54100A07BBF205F31193B2D3D'
    },
    'MaxLumi': {
        'MODEL': 'MaxLumi',
        'KEY': '3205089AF7E08108730ABFC237D94BE8',
        'IV': '09FE14E89ABE38654112885A43352D30'
    },
    'MaxLumi2': {
        'MODEL': 'MaxLumi2',
[... 226 lines cut ...]
if __name__ == '__main__':
    import sys
    if 2 <= len(sys.argv) <= 4:
        device_name = sys.argv[1]
        updateUpxPath = "update.upx" if len(sys.argv) == 2 else sys.argv[2]
        if len(sys.argv) == 4:
            decryptedPath = sys.argv[3]
        else:
            import os.path
            basename = os.path.basename(updateUpxPath)
            name, ext = os.path.splitext(basename)
            decryptedPath = name + '.zip' if ext == '.upx' else basename + '.zip'
        if device_name not in boox_strings.keys():
            print(f'The device "{device_name}" is not supported, or the name is wrong')
            print("Supported devices:")
            print(" ".join(sorted(boox_strings.keys())))
            sys.exit()
        decrypter = DeBooxUpx(**boox_strings[device_name])
        decrypter.deUpx(updateUpxPath, decryptedPath)
        print(f"Saved decrypted file to {decryptedPath}")
    else:
        print("Usage:\npython DeBooxUpdate.py <device name> [input file name [output file name]]")
        print("Supported devices: (those marked with suffix '-ru' are Russian models)")
        print(" ".join(sorted(boox_strings.keys())))
```

Hagb, 2023-11-22:
@RenateUSB Thank you! Could you please submit a PR?
Btw, `Steam` in `deUpxSteam` is a typo I made, which should be `Stream`. You could fix it together.

Iey4iej3, 2023-11-24:
Close this issue by https://github.com/Hagb/decryptBooxUpdateUpx/commit/d48b8cbc5cbaab4339840462da77ed7657d2357d

closed 2023-11-24 by Hagb