Hagb/decryptBooxUpdateUpx #76: strings for FLOWProMax 得到阅读器Max

cainiao5, 2023-12-11:
local
BnSyB10V3fkhpyZrxaRfdsvJDRnaHGv3ds0xTZ+N
setting
H1TjNwp/g9JehZevmKINGUB2Pafpy1ji7FVze0e64Hzg
upgrade
GCGXNX0J9NecC4xTAauGoGoIXBjDLDdEB1K81IjJf3we

http://firmware-volc.boox.com/51e407460864dc82528170fe9c1fb497/update.upx

RenateUSB, 2023-12-11:
What is the actual name of this model? "FLOWProMax", "FlowProMax" does not decode properly
`getprop ro.product.model`

RenateUSB, 2023-12-11:
What is the exact product name? "FLOWProMax" does not decode correctly.
`getprop ro.product.model`

cainiao5, 2023-12-12:
> What is the exact product name? "FLOWProMax" does not decode correctly. `getprop ro.product.model`
> 
> Hmm, that's strange because I get the same as you on the update:
> 
> ```
> Model:  FLOWProMax
> Finger: ONYX/FLOWProMax/FLOWProMax:10/2023-01-13_09-53_3.0.1_a77df95eb/7845:user/dev-keys
> Link:   http://firmware-volc.boox.com/51e407460864dc82528170fe9c1fb497/update.upx
> Size:   1522024975
> MD5:    51e407460864dc82528170fe9c1fb497
> ```

 the exact product name is FLOW Pro.Max,from getprop ro.product.model

RenateUSB, 2023-12-12:
That's very strange because 1) no other model has a space in its name, 2) it doesn't work.
```
C:\>BooxKeyConvert.py "FLOW Pro.Max" H1TjNwp/g9JehZevmKINGUB2Pafpy1ji7FVze0e64Hzg
H1TjNwp/g9JehZevmKINGUB2Pafpy1ji7FVze0e64Hzg
Key did not decode correctly for this model
```
I've tried a bunch of variants too.

cainiao5, 2023-12-12:
[Uploading OnyxOtaService.zip…]()

cainiao5, 2023-12-12:
[OnyxOtaService.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/13647554/OnyxOtaService.zip)

RenateUSB, 2023-12-12:
Thank you.

What does Settings > About phone > Model say?

cainiao5, 2023-12-12:
> Thank you.
> 
> What does Settings > About phone > Model say?

FLOW Pro.Max，I'll look for the machine to double-check.

RenateUSB, 2023-12-13:
Oh, I only had to try 8100 combinations to figure this out:
```
"FLOW.Pro Max"
6D0D6E2DBDF3D5D6C71B84E170E1E33A
11DFA3EAC57A65E2F9731DA5F278013C
```

closed 2023-12-13 by RenateUSB

cainiao5, 2023-12-13:
> Oh, I only had to try 8100 combinations to figure this out:
> 
> ```
> "FLOW.Pro Max"
> 6D0D6E2DBDF3D5D6C71B84E170E1E33A
> 11DFA3EAC57A65E2F9731DA5F278013C
> ```

i'm sorry,,,,,

RenateUSB, 2023-12-13:
No problem. I was just having a bit of fun seeing if I could figure it out.
It's merged now so it's in BooxKeys.csv for you to use.
My apologies if I closed this a bit quickly.