Hagb/decryptBooxUpdateUpx #79: Palma strings

carterprince, 2024-01-06:
```
desktop ~/.local/src/decryptBooxUpdateUpx $ python ota_jni.py ./libota_jni.so
ERROR:androidemu.internal.modules:=> Undefined external symbol:
ERROR:androidemu.internal.modules:=> Undefined external symbol: __cxa_finalize
ERROR:androidemu.internal.modules:=> Undefined external symbol: __register_atfork
ERROR:androidemu.internal.modules:=> Undefined external symbol: __cxa_atexit
ro.kernel.qemu was not found in system_properties dictionary.
libc.debug.malloc was not found in system_properties dictionary.
WARNING:root:File does not exist '/proc/stat'
WARNING:androidemu.internal.modules:libcrypto.so needed by ./libota_jni.so do not exist in vfs ExAndroidNativeEmu/vfs
WARNING:androidemu.internal.modules:libutils.so needed by ./libota_jni.so do not exist in vfs ExAndroidNativeEmu/vfs
----------------
STRING_SETTINGS tCXh+fW3GgMbB5Y6QT2rAT38WR3qyq4wGPqrQkOYnMG3
STRING_UPGRADE xibg8vjNHQmbw2PbkUJ8TH8Wh4gtBVw/OYGu+khjdqct
```

Did I do it right? Lol
Here's the .so, I'm zipping it because GitHub won't let me send a .so file: [libota_jni.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/13848738/libota_jni.zip)

carterprince, 2024-01-06:
Ah okay you have to run it through this?

```
desktop ~/.local/src/decryptBooxUpdateUpx $ python BooxKeyConvert.py Palma tCXh+fW3GgMbB5Y6QT2rAT38WR3qyq4wGPqrQkOYnMG3 xibg8vjNHQmbw2PbkUJ8TH8Wh4gtBVw/OYGu+khjdqct
tCXh+fW3GgMbB5Y6QT2rAT38WR3qyq4wGPqrQkOYnMG3
0E729B7296ACCB86AE5A6C15374C4D02

xibg8vjNHQmbw2PbkUJ8TH8Wh4gtBVw/OYGu+khjdqct
BF694808B49ACAF49E73334E3A4B1163
```

carterprince, 2024-01-06:
Was able to root my Palma by adding the above to `BooxKeys.csv`, so I assume they're correct.

Hagb, 2024-01-07:
Thanks for your keys. Merged.

closed 2024-01-07 by Hagb