Hagb/decryptBooxUpdateUpx #86: Request for Note Air 3 firmware package.

sellin82, 2024-01-19:
Request for Note Air 3 firmware package.

Note Air 3 model
I would like to download the firmware.

It doesn't download.

[http://data.onyx-international.cn/api/firmware/update?where={"buildNumber":0,"buildType":"user","deviceMAC":"","lang":"en_US","model":"NoteAir3","submodel":"","fingerprint":""}](http://data.onyx-international.cn/api/firmware/update?where=%7B%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22en_US%22,%22model%22:%22NoteAir3%22,%22submodel%22:%22%22,%22fingerprint%22:%22%22%7D)

This site address does not work.

RenateUSB, 2024-01-19:
Onyx is not currently offering any downloads on that model.
The website is returning 204, "No content".
On many browsers that may not be readily perceived.

Hagb, 2024-01-20:
refer to "2021-11-26 update" in https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389

sellin82, 2024-01-20:
> [#2(댓글)](https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389) 의 "2021-11-26 업데이트"를 참고하세요.[](https://github.com/Hagb/decryptBooxUpdateUpx/issues/2#issuecomment-704006389)

2021-11-26 update: Found that some firmware updates need "fingerprint" argument filled, such as fingerprint="Onyx/MaxLumi2/MaxLumi2:11/2021-11-18_04-40_3.2_6ab9d89216/31254:user/dev-keys" for MaxLumi2.
It is because Onyx now also releases cumulative updates. If all updates of a model are cumulative, the above url without fingerprint filled will return 204 No Content.

I don't know how to do this.

RenateUSB, 2024-01-20:
http://data.boox.com/api/1/firmware/update?where=%7B%22buildNumber%22%3A0%2C%22buildType%22%3A%22user%22%2C%22deviceMAC%22%3A%22%22%2C%22fingerprint%22%3A%22Onyx%2FNoteAir3%2FNoteAir3%3A11%2F2023-12-07_16-01_3.5_21f0f7717%2F3154%3Auser%2Frelease-keys%22%2C%22lang%22%3A%22en_US%22%2C%22model%22%3A%22NoteAir3%22%2C%22submodel%22%3A%22%22%7D

This returns 204 also.

sellin82, 2024-01-20:
The product I have is the Note X3, a Chinese product.

I am trying to obtain the global model Noteair3 model firmware and apply it to the Note x3 product.

If I cannot obtain the noteair3 firmware file, is it possible to apply the Noteair3C color product firmware file to the Note x3?

The difference between the Noteair3 and Noteair3c models is that the configurations such as RAM and CPU are the same, but only the screen appears to be different.

Wouldn't brick work?

RenateUSB, 2024-01-20:
Why didn't you say this about the NoteX3 before?

If you have a different device why don't you do something helpful for others, like posting the libota_jni.so

Hagb, 2024-01-20:
> I don't know how to do this.

Unfortunately, these may not be a "complete" Note Air 3 firmware package, but only cumulative firmware packages. A *cumulative* firmware package mean a package including only the updated and new content from the old firmware to new firmware, which doesn't include other parts of the firmware.

> If I cannot obtain the noteair3 firmware file, is it possible to apply the Noteair3C color product firmware file to the Note x3?

_In theory_, it is possible to obtain the full firmware file _from a Note Air 3 device_ indeed. But if you would apply it to your Note X3, I could not be sure that it would run properly.

> The difference between the Noteair3 and Noteair3c models is that the configurations such as RAM and CPU are the same, but only the screen appears to be different.

Even the RAM and CPU are the same, it is still possible that these two models have different hardware, probably including but not limited to the screen mentioned by you. It could lead to incompatible firmwares.

BTW if you don't trust ONYX's server in China, you can change it to the US or EU server (Settings->Server), or don't let it access to the Internet.

> If you have a different device why don't you do something helpful for others, like posting the libota_jni.so

@sellin82  Could you help with getting the keys of the firmware of Note X3? To do this, you could simply install and run [this app](https://github.com/Hagb/GetBooxUpxKeysApp/releases/download/v0.1.1/GetBooxUpxKeysApp-v0.1.1-release.apk) in you Note X3 and post its output to us.

RenateUSB, 2024-01-20:
Mmm, "cumulative" is correct, but I've always divided things between "full updates" (~1.5 GB) and "incremental updates" (~300 MB).

sellin82, 2024-01-20:
note Even if you want to increase the product key

GetBooxUpxKeysApp-v0.1.1-release.apk
After downloading, the app is installed.
It does not run and stops.

That's why I can't upload a photo of my key information.

RenateUSB, 2024-01-20:
> That's why I can't upload a photo of my key information.

Ok, but could you upload a copy of /system/lib64/libota_jni.so

sellin82, 2024-01-21:
/system/lib64/libota_jni.so

Where is this path?

note I'm looking for it in the X3 product folder, but it's not there??

sellin82, 2024-01-21:
[libota_jni.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/14000048/libota_jni.zip)

Note x3
It's a file.

RenateUSB, 2024-01-21:
Thanks. The keys are in the PR.

sellin82, 2024-01-22:
Is there any way to obtain the Noteair3 product firmware?

Hagb, 2024-01-22:
> Is there any way to obtain the Noteair3 product firmware?

There is probably no such a complete firmware package on the internet. You can try to get in touch with NoteAir3 owners who might be willing to give you their firmware. But please notice that flashing it into your NoteX3 may break your device.

May I ask what features you want so much are in the firmware of NoteAir3 but not in the firmware of NoteX3?

RenateUSB, 2024-02-16:
I don't think there is anything anyone here can do for you.
Keeping checking for an update to NoteAir3.

closed 2024-02-16 by RenateUSB

KingOfSpades, 2024-11-04:
FYI the URL now seems to work!

RenateUSB, 2024-11-04:
```
Model:  NoteX3 (Chinese)
Finger: Onyx/NoteX3/NoteX3:11/2024-08-12_14-40_3.5.4_c1328775d/5830:user/release-keys
Link:   http://firmware.boox.com/c3acd1c5ca381c748c2786030002758c/update.upx
Size:   1,767,086,849
MD5:    c3acd1c5ca381c748c2786030002758c
```