Hagb/decryptBooxUpdateUpx #88: Requesting the key, I found the firmware but don't know how to get the password, can you help me? Thank you so much

TiaoXinMa, 2024-01-29:
http://data.onyx-international.cn/api/firmware/update?where={%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22zh_CN%22,%22model%22:%22SP_PokeL%22,%22submodel%22:%22%22,%22fingerprint%22:%22%22}
This is my firmware address，it is SuperStar (chaoxing) verison，model is SP_POKEL
it's not in the list
My device stopped turning on because I tried to use RKDevTool to modify the system.
Please help me unpack it. I hope I can extract the correct system files and repair the device. Thank you very much.
[image]

RenateUSB, 2024-01-30:
Can you access /system in recovery? You'll probably have to mount it.
Without a copy of /system/priv-app/OnyxOtaService/OnyxOtaService.apk we can't do much.
If you can use Rockchip tools to copy the entire system partition we can probably find that apk.
The  update is encrypted.
```
Model:  SP_PokeL
Finger: Onyx/SP_PokeL/SP_PokeL:4.4.4/2019-04-30_20-20_super-star_cde7b0b/828:user/dev-keys
Link:   http://onyx-firmware.oss-cn-shenzhen.aliyuncs.com/1bb3b432698e18fdb1fb97aa5f347e7d.upx
Size:   304,028,566
MD5:    1bb3b432698e18fdb1fb97aa5f347e7d
```

TiaoXinMa, 2024-01-30:
I extracted the system partition from POKEL that others can run normally. These are all the priv-apps. Which one is needed?
[image]

TiaoXinMa, 2024-01-30:
https://drive.google.com/file/d/18GzRCcsqv-6vFQ2rz8RqWWyjPCfneSpR/view?usp=drive_link 
This is the extracted system

RenateUSB, 2024-01-30:
That isn't public, I can't view it.

TiaoXinMa, 2024-01-31:
https://drive.google.com/file/d/18GzRCcsqv-6vFQ2rz8RqWWyjPCfneSpR/view?usp=sharing
Sorry, I use Google Cloud Drive less, but it should be fine now, thank you very much

RenateUSB, 2024-01-31:
Oh, it was in /system/app/OnyxOtaService.apk
`SP_PokeL,SP_PokeL,6E01CB5A77A96F675561AE7CB7CAF6AC,DEAEF764C4290B4039EA7DDC2110A7B2`
There's a PR in for this.

TiaoXinMa, 2024-01-31:
Thank you very much. You are really a very enthusiastic and kind person. Thanks to your help, the file can be decrypted normally.

closed 2024-01-31 by RenateUSB