Hagb/decryptBooxUpdateUpx #97: Is there a key for Boox Note Plus, or Note+?

astroHaoPeng, 2024-05-16:
According to the official specs, `Boox Note Plus` seems not much different from `Boox Note`.

https://onyxboox.com/boox_noteplus
https://onyxboox.com/boox_note

Did anyone confirm they have the same key? Thanks. 

PS. They are obviously different from `Boox Note Pro`: https://onyxboox.com/boox_notepro

RenateUSB, 2024-05-18:
Undoubtedly they have different keys because they are different models.
You need to find sombody who owns a NotePlus.
Oh, this is a very old device.

RenateUSB, 2024-08-02:
These are all old products that are only listed in https://onyxboox.com/whatever
The new products are all under https://shop.boox.com/products/whatever

RenateUSB, 2025-11-06:
I believe that the Note Plus is the same hardware with a bigger battery.