Hagb/decryptBooxUpdateUpx #98: [New .UPX V3.5.2] How can I decrypt it?

xulubin, 2024-05-25:
Excuse me, recently I wanted to root my device, and I was lucky enough to find it here. First of all, thank the developers for sharing.  I couldn't find the corresponding key in the latest BooxKeys.csv, and tried [GetBooxUpxKeysApp-v0.1.1-release.apk](https://github.com/Hagb/GetBooxUpxKeysApp/releases) to get it, but.apk always flashed back after running, and it didn't work.  Also tried to extract libot_jni.so file, I do not know if there is a problem with my configuration environment or a new file update: Follow [algorithm-zh_cn.md](https://github.com/Hagb/decryptBooxUpdateUpx/blob/master/algorithm-zh_cn.md) related instructions to operate, you can get the results of the.so file in the test folder in the repository, but I extracted the .so file myself can not calculate the results.  Therefore, I hope there are kind-hearted people who can help me, thank you very much! 

PS:
[update.upx](http://data.onyx-international.cn/api/firmware/update?where={%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22zh_CN%22,%22model%22:%22NoteX3Pro%22,%22submodel%22:%22%22,%22fingerprint%22:%22Onyx/NoteX3Pro/NoteX3Pro:11/2024-05-15_18-13_3.5.2_50da13250/4785:user/release-keys%22})
V3.5.2
"buildDisplayId": "2024-05-15_18-13_3.5.2_50da13250"
"model": "NoteX3Pro"

[getfromboox.zip](https://github.com/Hagb/decryptBooxUpdateUpx/files/15444616/getfromboox.zip)
the getfromboox.zip contains OnyxOtaService.apk and libota_jni.so（psw：123）
Thanks！

RenateUSB, 2024-05-25:
It's added to BooxKeys.csv now.

closed 2024-05-25 by RenateUSB

xulubin, 2024-05-26:
It's unbelievable! Your efficiency is too amazing! Excellent! Thank you very much!