Hagb/decryptBooxUpdateUpx #104: Key for JDread required

SkinoAsu, 2024-07-20:
It's a unique device that boox customized it for JD (Chinese online shopping website)
It do not contain OnyxOtaService.apk in /system/priv-app/, but integrated the ota function in their own app jdread .
I pulled it from the device and then disassembled the apk (attached below). It's quite different between the common OnyxOtaService.apk where can I get the key?

https://1drv.ms/u/c/c812adfd1101b2bc/EZxiFeQO9ZFBvUwI1rjjDOwB5SwWmvvK4F4-j6iOCYo2Lw?e=iQRJmb

it's too big, I have to use onedrive to share

RenateUSB, 2024-07-20:
Which device is this? Can you `getprop | grep finger`
If this is the Poke5 (Chinese) and not the Poke5P (English) we already have the keys.

SkinoAsu, 2024-07-21:
> Which device is this? Can you `getprop | grep finger` If this is the Poke5 (Chinese) and not the Poke5P (English) we already have the keys.
> 
> There is a bunch of OTA stuff in here, but not what we are looking for. There is somewhere the com.onyx.android.onyxotaservice package.

getprop | grep finger does not output anything.
JDread runs customized boox system, which is vary different from others.
I can't get infomation from getprop, but I'm able to pull build.prop(attached below) from /system, I've already obtained root access on the device. 

the string ro.build.fingerprint seems to be Onyx/JDRead/JDRead:4.4.2/2020-12-16_02-10_jd_b94ff6b/1561:user/dev-keys

I can't find anything related to com.onyx.android.onyxotaservice package in /system/priv-app/ also /system/app/, but I would do further rechearch by pulling the whole /system image.

contents of build.prop below

```
# begin build properties
# autogenerated by buildinfo.sh
ro.build.id=2020-12-16_02-10_jd_b94ff6b
ro.build.display.id=2020-12-16_02-10_jd_b94ff6b
ro.build.version.incremental=1561
ro.build.version.sdk=19
ro.build.version.codename=REL
ro.build.version.release=4.4.2
ro.build.date=2020年 12月 16日 星期三 02:11:07 CST
ro.build.date.utc=1608055867
ro.build.type=user
ro.build.user=jenkins
ro.build.host=Onyx-BuildServ
ro.build.tags=dev-keys
ro.product.model=JDRead
ro.product.brand=Onyx
ro.product.name=JDRead
ro.product.device=JDRead
ro.product.board=JDRead
ro.product.cpu.abi=armeabi-v7a
ro.product.cpu.abi2=armeabi
ro.product.manufacturer=Onyx
ro.product.locale.language=zh
ro.product.locale.region=CN
ro.wifi.channels=
ro.board.platform=imx6
# ro.build.product is obsolete; use ro.product.device
ro.build.product=JDRead
# Do not try to parse ro.build.description or .fingerprint
ro.build.description=JDRead-user 4.4.2 2020-12-16_02-10_jd_b94ff6b 1561 dev-keys
ro.build.fingerprint=Onyx/JDRead/JDRead:4.4.2/2020-12-16_02-10_jd_b94ff6b/1561:user/dev-keys
ro.build.characteristics=tablet
# end build properties

#
# ADDITIONAL_BUILD_PROPERTIES
#
ro.carrier=unknown
ro.config.notification_sound=OnTheHunt.ogg
ro.config.alarm_alert=Alarm_Classic.ogg
dalvik.vm.heapstartsize=8m
dalvik.vm.heapgrowthlimit=64m
dalvik.vm.heapsize=384m
dalvik.vm.heaptargetutilization=0.75
dalvik.vm.heapminfree=512k
dalvik.vm.heapmaxfree=8m
persist.sys.strictmode.visual=0
persist.sys.strictmode.disable=1
ro.sf.lcd_density=300
persist.sys.timezone=Asia/Shanghai
persist.sys.dalvik.vm.lib=libdvm.so
dalvik.vm.dexopt-flags=m=y
net.bt.name=Android
dalvik.vm.stack-trace-file=/data/anr/traces.txt
```

RenateUSB, 2024-07-21:
JDRead is old. It's Android 4.4.2 The last update I see on the regular Onyx channel is from 2018. 
http://onyx-firmware.oss-cn-shenzhen.aliyuncs.com/5a031af9033c0051c274defbbe534391.zip
That's a regular unencrypted zip file, but the individual files are encrypted.

The file that you posted seems to be using the newer update.upx format.

SkinoAsu, 2024-07-21:
> JDRead is old. It's Android 4.4.2 The last update I see on the regular Onyx channel is from 2018. http://onyx-firmware.oss-cn-shenzhen.aliyuncs.com/5a031af9033c0051c274defbbe534391.zip That's a [...]

Interesting, I have a firmware package that directly copied from the device storage which is .upx format the firmware version is 202012160210.But I guess the most difference of firmwares is JDRead-release.apk

Seems that boox have changed their update format after 2018.

Tanks for helping me to get the unencrypted firmware

I will provide the .upx package here

https://1drv.ms/u/c/c812adfd1101b2bc/EZHeG8ZzXZhOr0_0xR_I3nYBGayLydJ1XtI1YcZx3rbU9Q?e=DtKrVy

RenateUSB, 2024-07-21:
We still haven't found the keys. Is there a libota_jni.so (or similar) somewhere?

SkinoAsu, 2024-07-21:
> We still haven't found the keys. Is there a libota_jni.so (or similar) somewhere?

I'm still working on it. But find nothing like that but I can find something related to encryption or ota, they are in .smali format but I know nothing about smali. I'm a complete noob in reverse engineering android apps

Could you offer me some infomation about com.onyx.android.onyxotaservice package? size?
some smali files do contain this package name and com.onyx.android.onyxotaservice.OtaInfoActivity
I strongly doubt it that com.onyx.android.onyxotaservice package is in someware but with a different name

SkinoAsu, 2024-07-21:
> We still haven't found the keys. Is there a libota_jni.so (or similar) somewhere?

I just find something might related
there is a file called otacerts.zip(attached) under /system/etc/security/. After unzip that, I get testkey.x509.pem

[otacerts.zip](https://github.com/user-attachments/files/16324823/otacerts.zip)

RenateUSB, 2024-07-21:
Nope, that's not it. Try:
```
cd /system/framework
grep onyxota *
```

SkinoAsu, 2024-07-21:
> Nope, that's not it. Try:
> 
> ```
> cd /system/priv-app
> grep -r onyxota *
> ```
> 
> Edit: Oops, it is an Activity.

grep -r onyxota * do not output anything.
So,I  tried pm list packages then it returns this
```
package:com.android.soundrecorder
package:com.onyx.android.production.test
package:com.android.defcontainer
package:android
package:com.android.settings
package:com.onyx.jdread
package:cn.modificator.launcher
package:com.speedsoftware.rootexplorer
package:org.koreader.launcher
package:com.android.systemui
package:com.android.bluetooth
package:com.shere.assistivetouch
package:com.android.keychain
package:com.android.inputdevices
package:com.android.packageinstaller
package:com.android.browser
package:com.eusoft.eudic
package:eu.chainfire.supersu
package:com.wsw.cospa
package:com.android.dialer
package:com.android.backupconfirm
package:com.android.providers.settings
package:com.kadbbz.smartcleaner
package:com.android.keyguard
package:com.iflytek.inputmethod
package:com.senventise.jdread1
package:com.android.provision
package:com.onyx.pinyinime
**package:com.onyx.android.update**
package:com.android.pacprocessor
package:com.android.providers.media
package:com.android.shell
package:com.android.certinstaller
```
**package:com.onyx.android.update**
should this work?

RenateUSB, 2024-07-21:
Maybe. Something is screwy here.

Run a logcat in one window. Run an ADB shell in another.
am start -n com.onyx.android.onyxotaservice/.OtaInfoActivity
am start -n com.onyx.android.onyxotaservice/.OtaInfoActivity -e updatePath sillypath
Any good messages?

SkinoAsu, 2024-07-21:
> Maybe. Something is screwy here.
> 
> Run a logcat in one window. Run an ADB shell in another.
> 
> ```
> am start -n com.onyx.android.onyxotaservice/.OtaInfoActivity
> am start -n com.onyx.android.onyxotaservice/.OtaInfoActivity -e updatePath sillypath
> ```
> 
> Any good messages?

Unfortunately, I bricked my device by installing busybox on it just now
I have to wait until oct to grab another one from my parents' house 
I guess, I can either find a way to unbrick it or just going on with exist information

RenateUSB, 2024-07-21:
December???

SkinoAsu, 2024-07-21:
> December???

Yes, my plan to visit my parents is placed in December, Then I'll take another device back as well as my emmc programmer

Before I bricked it, I'm able to backup my /system(mmcblk0p5) for analysis and restore, image is uploaded to one drive
Considering it became to refuse booting up, the only option to restore backup is programmer
It also boots into recovery mode with a blank screen, but my computer can't recognize it (code 43).
So, it's unable to unbrick with recovery.

https://1drv.ms/u/c/c812adfd1101b2bc/ETqvddGU3ddIhegCSGlC6jQB889JBAwKTIjPnO51Gm9QEA?e=nEiUKU

RenateUSB, 2024-07-22:
I've looked a lot and come to the conclusion that that stuff was in an OnyxOtaService.apk and somebody deleted it when they rooted this.

SkinoAsu, 2024-07-22:
> I've looked a lot and come to the conclusion that that stuff was in an OnyxOtaService.apk and somebody deleted it when they rooted this.

No actually
Ota function works fine after I root it ，because the update package is a full package. After putting .upx in /sdcard/ and install it,
the device will do factory reset and overwrite the system parturition.

My suspicions is that the jdread have taken over the ota function because the original system is designed to launch jdread after boot, all the system settings/operation/function(including ota) is built in jdread. Users are not allowed to quit jdread, so the unpacking function should be integrated into jdread.

RenateUSB, 2024-07-22:
> No actually Ota function works fine after I root it ，because the update package is a full package. After putting .upx in /sdcard/ and install it, the device will do factory reset and overwrite the system parturition.

Mmm, I'm not convinced. Multiple packages contain `startFirmwareUpdate()` which basically does those start commands above. I don't find hide nor hair of `com.onyx.android.onyxotaservice` anywhere.