Hagb/decryptBooxUpdateUpx #107: Key of GALILEO

xtotdam, 2024-10-04:
Hi! I managed to obtain keys for Galileo, which is sold in Russia.

http://onyx-boox.ru/boox_galileo
http://onyx-boox.ru/files/galileo/update.upx

```
"MODEL" = "GALILEO",
"STRING_SETTINGS" = "pjkfzKJZ9ZQNSuh9hMmAHO9OrG2PedNpiMNASMunHgoV",
"STRING_UPGRADE" = "0Tocz9NeiODK9jtrwopAgEf5Dr5Gpr3biK32jAjYBLT3"
# fingerprint: Onyx/GALILEO/GALILEO:11/2024-05-28_15-32_3.3.2-mc_1a6a21edc/8486:user/dev-keys
```

```
pjkfzKJZ9ZQNSuh9hMmAHO9OrG2PedNpiMNASMunHgoV
F303E7819D87666785B5CEE6C172201D

0Tocz9NeiODK9jtrwopAgEf5Dr5Gpr3biK32jAjYBLT3
103040EE4F225D67A607B25337E787B9
```

I managed to successfully decrypt update.upx and now wonder what to do with a big payload.bin file))

RenateUSB, 2024-10-04:
I get a 403 error on http://onyx-boox.ru/files/galileo/update.upx

xtotdam, 2024-10-05:
My bad, I just copied link from my browser. https://onyx-boox.ru/files/galileo/update.upx works good for me, http link doesn't work either.

RenateUSB, 2024-10-05:
Well, it must be geo-blocked or else it's using some cookie that you have that I don't.
You can try it with curl (like I did).
`curl https://onyx-boox.ru/files/galileo/update.upx -o update.upx`

xtotdam, 2024-10-07:
I doubt it is geoblocked because I tried the link from different countries via vpn, and I came to this site without any cookies as well:) I think it was just a wrong protocol in my first message.

RenateUSB, 2024-10-07:
No, I am still getting 403 using https on either browser or curl.
```
curl https://onyx-boox.ru/files/galileo/update.upx
<html>
<head><title>403 Forbidden</title></head>
<body>
<center><h1>403 Forbidden</h1></center>
<hr><center>nginx</center>
</body>
</html>
```

xtotdam, 2024-10-07:
Must be geoblock then, I tried another country and got 403 too. Thing is I am in Russia, so it just works (c)

RenateUSB, 2024-10-07:
I just wanted to double check that the keys were working.
Can you chop me off 64k of that update.upx please?
`dd if=update.upx of=short.img count=128`

xtotdam, 2024-10-07:
No problem. Github doesnt like .img files, so I renamed it
[short.img.txt](https://github.com/user-attachments/files/17277932/short.img.txt)

RenateUSB, 2024-10-07:
Thank you! It has been added.

closed 2024-10-07 by RenateUSB