Hagb/decryptBooxUpdateUpx #112: Failed to verify whole-file signature" error after sideloading decrypted firmware on Onyx Boox Poke5 (zh_CN)

wonnew1207, 2024-11-15:
Hello,
I am using an Onyx Boox Poke5 (zh_CN) model. My device originally had the 3.5.3 Chinese firmware installed, and I attempted to switch to the Poke5P (en_US) 3.5.3 global firmware. However, during this process, my device entered a bootloop.
To resolve this, I obtained the 3.5.4 firmware for Poke5 (zh_CN) and used decryptBooxUpdateUpx.py to decrypt the update.upx file into an update.zip file.
After that, I accessed Recovery Mode -> Apply update from ADB, and tried to sideload the decrypted update.zip file. However, I encountered a "failed to verify whole-file signature" error during the flashing process.
Do you have any insights or suggestions on how to resolve this issue?
Thank you in advance for your help!

RenateUSB, 2024-11-15:
>Do you have any insights or suggestions on how to resolve this issue?

Yes, but this would be better handled in the forums of mobileread.com
3.5.4 for Poke5P is also available.

wonnew1207, 2024-11-18:
> > 이 문제를 해결하는 방법에 대한 통찰력이나 제안이 있나요?
> 
> 네, 하지만 이 문제는 mobileread.com 포럼에서 처리하는 게 더 좋을 겁니다 . Poke5P 3.5.4도 사용 가능합니다.

Hello Renata,
I already knew about you through the mobileread.com site. I am very happy that you replied to me. Thank you so much.
I have some personal questions I would like to ask you. Would it be possible to communicate via email? My email address is ggdd1130@gmail.com.

wonnew1207, 2024-11-18:
Hello Renate,
(I am currently waiting for the comment to be approvedI am very happy that you replied to me. Thank you so much.
Following your suggestion, I have signed up for the MobileRead forum and left a comment. (I am currently waiting for the comment to be approved.)

RenateUSB, 2024-11-19:
Conversation continued off-site.

closed 2024-11-19 by RenateUSB