Hagb/decryptBooxUpdateUpx #118: Require firmware link of Go6

mio-19, 2025-02-16:
My Go6 is currently on 2025-02-11_23-36_v4.0-beta4_a390c414ef. How do I extract fingerprint from the device and construct a firmware download link?

RenateUSB, 2025-02-16:
To get a link to a beta you must supply a WiFi MAC of a device that is subscribed to the beta channel.
```
Model:  Go6 (English)
Finger: Onyx/Go6/Go6:11/2025-02-11_23-36_v4.0-beta4_a390c414ef/48880:user/release-keys
Link:   http://firmware-us.boox.com/cb6afbde5365d8fc8ab86ab94490aad4/update.upx
Size:   1,752,091,374
MD5:    cb6afbde5365d8fc8ab86ab94490aad4
```

mio-19, 2025-02-17:
> To get a link to a beta you must supply a WiFi MAC of a device that is subscribed to the beta channel.
>
> ```
> Model:  Go6 (English)
> Finger: [...]

Could a process for doing this be documented in a markdown file within this repo?

mio-19, 2025-02-17:
Thank you so much Renate

closed 2025-03-06 by RenateUSB