Hagb/decryptBooxUpdateUpx #120: Keys for Darwin 7

fr33mang, 2025-04-01:
OTA app located in `system/app/OnyxOtaService.apk`
```
<string name="settings">FnEWuD7/sqZilCGGgnNIbHXAxjYh5GBMfBA7zXwz6nYN</string>
<string name="upgrade">GAZkzE6Mw6C0nUv2g+ON5/+61zkIucwIc849mcwUZAuA</string>
```

fr33mang, 2025-04-01:
> Thanks!
> 
> But it doesn't decode as "Darwin7". Could you check what variant of the name they are using? Look for things in `getprop` or `getprop ro.product.model`. I tried "Darwin7-ru" and various capitalizations.

MC_DARWIN7,MC_DARWIN7,9272F672593B96AE388849C30EC9272F,7EEF6EF4C34231E4E16A4C688DBF7C4B

fr33mang, 2025-04-01:
Looks like bootloader is locked on my reader so I cannot see my device in `fastboot` after `adb reboot bootloader`
But thank you anyway, I was able to research the device firmware better using your tools

RenateUSB, 2025-04-02:
Thanks for the update on the model name. The keys have been added.

Do you have a link for a Darwin7 update?

You don't need "bootloader" (i.e. abl fastboot), you need EDL mode (i.e. PBL), which you have.

Your fastboot could have a non-standard USB VID/PID. That needs investigating.

fr33mang, 2025-04-02:
> Do you have a link for a Darwin7 update?

@RenateUSB, you can find different versions [here](https://onyxboox.com/support/boox_darwin7)

fr33mang, 2025-04-02:
> You don't need "bootloader" (i.e. abl fastboot), you need EDL mode (i.e. PBL), which you have. Your fastboot could have a non-standard USB VID/PID. That needs investigating.

Thank you! I'll try EDL

RenateUSB, 2025-04-02:
Hmm, that link on the page gives me 404. Do you get that too?

fr33mang, 2025-04-02:
> Hmm, that link on the page gives me 404. Do you get that too?

oh same 404 for me
there is another one: https://onyx-boox.ru/support/boox_darwin7
works for me, but that software for russian version I suppose

RenateUSB, 2025-04-02:
Thanks. I just wanted to confirm that the decrypt works. I only downloaded 64km but that's good enough.

closed 2025-05-11 by RenateUSB