Hagb/decryptBooxUpdateUpx #121: Boox Note Max keys

VicVer09, 2025-05-20:
Hi, which keys should be used for the Boox Note Max.

I noticed there's a few choices in the csv

```
Max2,Max2,7B71BD797D7E3721D69F4449E12B4C22,38B2BB851DAF20E1BE6E7E3F349CABA0
Max2Pro,Max2Pro,8BEAEE4D037DAC8E2DB18458CDFA4A53,706038B3497224CCC2B81572EACAA780
Max3,Max3,0C3FF83E57334391A2D4171E99EAC1A9,982877D54100A07BBF205F31193B2D3D
MaxLumi,MaxLumi,3205089AF7E08108730ABFC237D94BE8,09FE14E89ABE38654112885A43352D30
MaxLumi2,MaxLumi2,25F86689E5FE3DEB2856983CCF21FB6D,14E98D0F357F7087D27D22406D7291E0
Note,Note,1D7A5202269D384B72F348CE25D55120,A0828A338A06FF2CD3618B219D3E30E2
Note2,Note2,47363CF9416ADE0231B61D62DD8F4539,462B820F6C4B3BC72EE48FDFF0FA3E39
Note3,Note3,098C863EB6E1F8DC96720031B32EA200,EFC95EF6B02442EE518F2D2BD5594250
Note5Plus,Note5Plus,9377320B7032EDC0194DBA82AD9DDCB0,9B7422F02E481B143030F24F5F3736B0
NoteAir,NoteAir,AED5E8AE8AB08319EC791849310171C2,F62715B99809212131696C2FD7C6F75D
NoteAir2,NoteAir2,856C9D4ADCE8015F272357F5DDF08B71,879D69054D57C76152CF272A0C121006
NoteAir2P,NoteAir2P,1276ECC6636DA6AFCEF5D7F300E5C915,1F6CF46EC279E61EDB7FC150C3471972
NoteAir3,NoteAir3,54F4360CD7465F72D31957D9DAF017C7,40FE355773181D3B8D0B3E41B96D6998
NoteAir3C,NoteAir3C,4DEAED7F843CDCEDB384E6FC468C540E,964E8856F922178810AEC06E0D363512
NoteAir4C,NoteAir4C,CF26B9FD1C5F74A8170C24D389F9A92D,3BC02F669B2C772CC3F9A583F1FC3263
NotePro,NotePro,F72D46836B5B1A0D1CB5BB27C6895A4F,2652833CDDC4C4CDD3BEC6F8D6AD3914
NoteS,NoteS,0B82F60DB060E404F752E9B91F0BE28B,363A003F2C54F1DA3412F7A9CC7DFDBE
NoteX,NoteX,8CC1464D7009076D571D4FF249F47B3D,2E3D8332638FED444BA24D5462C12046
NoteX2,NoteX2,00D996A8B45734B9E7943C0FF3333646,AE9ACFDF9C871C2B9458B2F2245035E0
NoteX3,NoteX3,40B1F5E5773AA90DC41ED62AA34EEF6D,C41B0F1CD0E88A0CC6CA22E1429AC4A3
NoteX3S,NoteX3S,37BD0706BCBCA53265083A292966FD4D,6EB0274D85707BA9729580E49D0890AA
NoteX3Pro,NoteX3Pro,7002375BDC1F2FE5FD29C3A0D3470883,C5334CAA3B27498621CFB7744F578797
Note_YDT,Note_YDT,A5BEEB144025D73FD90971E4906232BA,18870C4F426A494320868606DA2091BB
```

none of them look quite right. based on the official model i would assume we need `NoteMax`:

```
$ adb shell getprop ro.product.device
NoteMax
$ adb shell getprop ro.product.model
NoteMax
```

any help greatly appreciated

RenateUSB, 2025-05-20:
Yes, "NoteMax" is what they're calling it and it appears that we don't have the keys for it yet

If you could pull /system/lib64/libota_jni.so off your device and post it, we can fix that.
You may have to copy it first to /sdcard and then adb pull it.

RenateUSB, 2025-05-20:
Sorry, you converted that to Unicode and it messed up everything.
Just zip it.

VicVer09, 2025-05-20:
ah thats much easier:

[libota_jni.zip](https://github.com/user-attachments/files/20354367/libota_jni.zip)

RenateUSB, 2025-05-20:
You're killing me.
That's a zipped up version of the file after it was converted to Unicode.

The file is supposed to be 28000 or so.
It's twice as big because it's been mangled to death.
Check the file on your device.

VicVer09, 2025-05-20:
my apologies i am not an android dev. the unicode was a result of the command recommended in CONTRIBUTING.md: `adb pull /system/lib64/libota_jni.so` from a windows pc. i was able to get around it with copying it to sd first

```
adb shell cp /system/lib64/libota_jni.so /sdcard/libota_jni_fixed.so
adb pull /sdcard/libota_jni_fixed.so .\libota_jni_fixed.so
```

here is the 28000 file:

[libota_jni_fixed.zip](https://github.com/user-attachments/files/20354683/libota_jni_fixed.zip)

let me know if this is the correct format, and i appreciate your patience with this