Hagb/decryptBooxUpdateUpx #122: Boox Go Color7 (Gen II) - GoColor7_2

Lightsockie, 2025-05-28:
Just got a Go Color7 (Gen II) to replace another boox I recently busted (but previously had rooted as well).

```
$ adb shell getprop ro.product.model      
GoColor7_2
```

I'm trying to be careful that I don't assume too much, so was going through the process to verify the keys matched whats already defined in `BooxKeys.csv`. I saw that my internal product model isn't _quite_ `GoColor7`. I also see that my current build as `2025-04-30_23-39_v4.0-factory-0430_a682e4dbc`, but the latest `update.upx` I can get the magic URL to feed me when I use `GoColor7` is `"2025-03-26_18-59_v4.0-rel_2aa96eb60"`. I can't seem to get the magic URL to give me any results for `GoColor7_2`, and nothing I drop into the `submodel` field seems to do anything either.

All leading me to worry this is a distinct model from what's already tracked. 

---

I'm having trouble getting the `ota_jni.py` script to play nicely. I see that its pretty old, and made a virtualenv for `py1.11` and installed the `requirements.txt` from `ExAndroidNativeEmu`, but was getting an error around `pystr = emulator.java_vm.jni_env.get_local_reference(r).value.get_py_string()`. I saw that this string is just used to print something for debugging, though, so I commented it out, and here's the full error:

```
$ python ota_jni.py ./libota_jni.so
ERROR:androidemu.internal.modules:=> Undefined external symbol: 
ERROR:androidemu.internal.modules:=> Undefined external symbol: __cxa_finalize
ERROR:androidemu.internal.modules:=> Undefined external symbol: __register_atfork
ERROR:androidemu.internal.modules:=> Undefined external symbol: __cxa_atexit
ro.kernel.qemu was not found in system_properties dictionary.
libc.debug.malloc was not found in system_properties dictionary.
WARNING:root:File does not exist '/proc/stat'
WARNING:androidemu.internal.modules:libcrypto.so needed by ./libota_jni.so do not exist in vfs ExAndroidNativeEmu/vfs
WARNING:androidemu.internal.modules:libutils.so needed by ./libota_jni.so do not exist in vfs ExAndroidNativeEmu/vfs
ERROR:androidemu.internal.modules:=> Undefined external symbol: AES_set_encrypt_key
ERROR:androidemu.internal.modules:=> Undefined external symbol: AES_cfb128_encrypt
----------------
ERROR:root:Unable to find symbol 'Java_com_onyx_android_onyxotaservice_RsaUtil_nativeGetSecretKey' in module './libota_jni.so'.
ERROR:root:Unable to find symbol 'Java_com_onyx_android_onyxotaservice_RsaUtil_nativeGetIvParameter' in module './libota_jni.so'.
```

So anyway, I did give it an earnest effort but turning over the `libota_jni.so` file so ya guys can take a look
https://files.catbox.moe/1k28qg.so

Also, maybe already well known now, but I also tried going the "fetch `/data/local/update.zip` route as well. But it seems like `su` (or `sudo` or `doas`) isn't in shell's path anymore (verified its not in `/bin/`, at least). So can't access that path

RenateUSB, 2025-05-29:
Thank you. The GoColor7_2 has been added.
There have been and will be changes in how the keys are stored.

Updates are often not available for new devices.
You'll have to wait a bit, keep trying.

closed 2025-05-31 by RenateUSB

Lightsockie, 2025-07-03:
New update came out! Just updating the thread to show I got it working, and the model is indeed `GoColor7_2` (so no need to mess with the `submodel` it seems).

```
$ curl http://data.onyx-international.cn/api/firmware/update\?where\=%7B%22buildNumber%22:0,%22buildType%22:%22user%22,%22deviceMAC%22:%22%22,%22lang%22:%22en_US%22,%22model%22:%22GoColor7_2%22,%22submodel%22:%22%22,%22fingerprint%22:%22%22%7D | jq -r '.downloadUrlList[]' | wget -i-
```
```
$ python DeBooxUpx.py GoColor7_2 update.upx
Saved decrypted file to update.zip
```
ez